pybamm-team/PyBaMM #481: Add handling of derivatives of derived variables

Scottmar93, 2019-06-27:
In SPMe we have explicit expressions for `phi_s`. We want to take gradients of `phi_s` to obtain the ohmic heating. However `phi_s` is not in the boundary conditions (and shouldn't be) so when we take the gradient the resulting symbol is of the wrong size.

Scottmar93, 2019-07-01:
add to boundary conditions on a case by case basis. this will not affect the model checks

TomTranter, 2019-07-03:
Can you explain how you evaluate the gradient please

TomTranter, 2019-07-03:
Is there a way to access gradients of variables easily such as 
```
phis_neg = pybamm.ProcessedVariable(model.variables['Negative electrode potential [V]'], solution.t, solution.y, mesh=mesh)
grad_phis = pybamm.Gradient(model.variables['Negative electrode potential [V]'])
grad_phis.evaluate(solution.t, x)
```

Scottmar93, 2019-07-05:
Sorry for missing this @TomTranter. Currently, we can only take gradients before we discretise the model. This is because `pybamm.grad` creates a `pybamm.Gradient` symbol in the expression tree which is then converted to a matrix when the model is discretised (which is dependent upon the type of spatial method that is employed e.g. finite volume). 

Therefore, the easiest way to get direct access to the gradient of a symbol would be to add it to the model.variables dictionary before discretising and solving the model in a similar manner to other output variables. The following should work for the DFN: 

```
import pybamm
import numpy as np

model = pybamm.lithium_ion.DFN()

phi_s_n = model.variables["Negative electrode potential"]
model.variables.update(
    {"Gradient of negative electrode potential": pybamm.grad(phi_s_n)}
)

geometry = model.default_geometry

param = model.default_parameter_values
param.process_model(model)
param.process_geometry(geometry)

mesh = pybamm.Mesh(geometry, model.default_submesh_types, model.default_var_pts)

# discretise model
disc = pybamm.Discretisation(mesh, model.default_spatial_methods)
disc.process_model(model)

# solve model
t_eval = np.linspace(0, 2, 100)
solution = model.default_solver.solve(model, t_eval)

model.variables = pybamm.post_process_variables(
    model.variables, solution.t, solution.y, disc.mesh
)

grad_phi_s_n = model.variables["Gradient of negative electrode potential"]
``` 

This won't work at the moment for the SPM and SPMe until this issue is resolved. Once this issue is resolved, it probably makes sense to add gradients of phi into the set of variables which are automatically generate by each submodel so that they are readily available without having to do the above.  

I do agree that it would be nice to have a nice way of taking gradients after solving the model so it would make sense for us to look into ways of adding this type of feature in the future.

TomTranter, 2019-07-05:
Hi @Scottmar93,
Thanks for the explanation - I think simplest would be to add gradients automatically as long as this doesn't add too much memory. The list of variables is already quite long :)

PR #529 (2019-07-11): Issue 481 add handling derivatives of derived variables
# Description
Added derivatives of derived variables. Gradients can now be taken of phi_e_n etc for SPMe. Additionally, taking gradients of subdomain variables has been extended to the 2+1D models. Ohmic heating terms in all the thermal models have also been added. 

Still failing 1 debug mode test for 2+1 models. 

Fixes #481 

## Type of change

Please delete options that are not relevant.

- [x] New feature (non-breaking change which adds functionality)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ pytho [... cut]
closed 2019-07-11 by Scottmar93