pybamm-team/PyBaMM #494: Running thermal_lithium_ion example doesn't find parameters

TomTranter, 2019-07-04:
**Describe the bug**
FileNotFoundError: [Errno 2] File b'C:\\Code\\PyBaMM\\examples\\scripts\\input\\parameters\\lithium-ion\\mcmb2528_lif6-in-ecdmc_lico2_parameters_Dualfoil.csv' does not exist: b'C:\\Code\\PyBaMM\\examples\\scripts\\input\\parameters\\lithium-ion\\mcmb2528_lif6-in-ecdmc_lico2_parameters_Dualfoil.csv'

**To Reproduce**
Steps to reproduce the behaviour:
check out master
run example

I think it needs these lines added like the other examples
import os
os.chdir(pybamm.__path__[0]+'/..')
or perhaps the base_battery_model should try to find a file in cwd and if not found go to the root

martinjrobins, 2019-07-04:
thanks for reporting this @TomTranter. 

Pybamm should automatically use the root directory where you installed it. you actually shouldn't need `os.chdir(pybamm.path[0]+'/..')` anymore. 

how did you install pybamm, was it using `pip install <path_to_pybamm>`?

can you run this and let me know what you get:
```python
import pybamm
print(pybamm.root_dir())
```

TomTranter, 2019-07-04:
I did use pip and have the code in the the directory ''c:\\code\\pybamm'' which is found by the root_dir() function. However, Line 27 of base_battery_model uses os.getcwd()
```
input_path = os.path.join(os.getcwd(), "input", "parameters", "lithium-ion")
```

martinjrobins, 2019-07-04:
ahhh, I see it now. @Scottmar93 just merged in a refactor of the models that introduced this. I'll change this back to `pybamm.root_dir()`

Scottmar93, 2019-07-05:
Sorry about this. Thank you for catching this Martin

PR #495 (2019-07-06): #494 change os.getcwd to pybamm.root_dir
# Description

Fixes #494 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that prove fix is effective or that feature wor [... cut]
closed 2019-07-06 by tinosulzer