pybamm-team/PyBaMM #498: Add further instructions for installing without -e

Scottmar93, 2019-07-05:
Jon had a small issue when installing PyBaMM. He installed without developers options. Import pybamm worked fine within the PyBaMM/ directory but no module `pybamm` was found in the subdirectories (e.g. PyBaMM/examples/). Solution was to add the pybamm directory to the path. Should this happen automatically or should we add additional instructions for this?

Scottmar93, 2019-07-05:
solution was: 
`export PYTHONPATH=$PYTHONPATH:/path/to/pybamm/`

jonchapman1, 2019-07-05:
ipython finds pybamm OK.
But Jupyter cannot find pybamm.
I noticed the same issue on Binder. When I run a notebook through there it cannot find pybamm.

martinjrobins, 2019-07-05:
solution requires adding this line in `setup.py`:

```python
packages=find_packages(),
```

I'll add this in

PR #504 (2019-07-06): Issue 498 find packages
# Description

Fixes #498 

also minor fix with path issues in lead acid parameters

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All tests pass: `$ python run-tests.py --unit`
- [x] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run-tests.py --quick`.

## Further checks: 

- [x] Code is commented, particularly in hard-to-understand areas
- [x] Tests added that pr [... cut]
closed 2019-07-06 by tinosulzer