pybamm-team/PyBaMM #513: Benchmarking

tinosulzer, 2019-07-07:
Now that we have multiple solvers and models, we could start to do some benchmarking comparing them. Maybe a nightly CRON job on travis to print to a readme file? e.g. something like

# Benchmarking

## Doyle-Fuller-Newman model

Scottmar93, 2019-07-07:
This seems a good idea. Is the hardware being used by CRON jobs a good representation laptop speed though? Even if not still good for relative speed of models.

tinosulzer, 2019-07-07:
Yeah good point, will be worth checking this. Ran all the latest tests on master in 200 seconds so it's not awful

tinosulzer, 2019-07-09:
I basically have this working. 
@martinjrobins - is it possible / do you know how to configure travis to print to a file and then push back to GitHub?

martinjrobins, 2019-07-10:
yes, it is. You need to generate a Github token, and then encrypt it using travis before you use it in the `.travis.yml`. You can then use the token when you try to push to the repository. Here is a fairly decent writeup:

https://medium.com/@preslavrachev/using-travis-for-secure-building-and-deployment-to-github-5a97afcac113

ikorotkin, 2019-07-10:
That benchmarking table example looks promising!

But I would expect that dae-cpp might be the slowest for ODEs (especially for relatively small systems), because Scipy and Scikits may employ a different, more efficient numerical method for ODEs.
And even for DAEs the manual solver can be slower, because by default it works like in "safe mode". But after fine-tuning (after tweaking some of the solver options), it can work faster. For example, by default the solver requests a new updated Jacobian every Newton iteration within the time step (for better stability and accuracy, for stiff problems). But this is not necessary in many cases and we can ask the solver (through the solver options) to re-use old Jacobians. For example MATLAB ode15s makes this decision automatically for users, but the manual solver doesn't. It's like a car with manual transmission.

It also highly depends on how you provide Jacobian for the solver. If it is not provided, the solver estimates it numerically, and this will the slowest part of the solution process. If PyBaMM estimates it numerically... idk, maybe the solver can do it faster, maybe not. So benchmarking would be very interesting. Subscribed to this issue : )

tinosulzer, 2019-07-10:
Ok, I think there are lots of advantages to using your solver anyway (as discussed on Friday) even if it isn't immediately faster - such as being able to make those kinds of decisions about the Jacobian.
I'm hoping that the benchmarking file will be an easy reference point to check which solver to use in which case

tinosulzer, 2020-02-24:
Closing here, will implement in the case-studies repository instead

closed 2020-02-24 by tinosulzer