pybamm-team/PyBaMM #514: Master branch failing

Scottmar93, 2019-07-08:
Master branch is failing related to scikits-FEM i think following pull request #487.

error is
```
File "/home/travis/build/pybamm-team/PyBaMM/pybamm/meshes/scikit_fem_submeshes.py", line 93, in __init__
    lambda y, z: self.on_boundary(y, z, tabs["negative"])
  File "/home/travis/virtualenv/python3.6.7/lib/python3.6/site-packages/skfem/assembly/global_basis/global_basis.py", line 168, in get_dofs
    facets = self.mesh.facets_satisfying(facets)
  File "/home/travis/virtualenv/python3.6.7/lib/python3.6/site-packages/skfem/mesh/mesh.py", line 360, in facets_satisfying
    return np.nonzero(test(np.array(midp)))[0]
TypeError: <lambda>() missing 1 required positional argument: 'z'
```

martinjrobins, 2019-07-08:
the lambda has two arguments, y and z, but the code in scikits fem calls the lambda with a single np.ndarray. @rtimms: do you know why this suddenly doesn't work?

rtimms, 2019-07-08:
@martinjrobins looks like scikit changed the interface for the method `facets_satisfying` from being coords x,y to an array of coords. have fixed this now. however, running `run-test.py --unit` all the tests pass, but get a shape error on `test_spatial_methods.test_scikit_finite_element.TestScikitFiniteElement` when running `run-tests.py --quick`. running the test on its own is also fine. any ideas what might be causing this?

tinosulzer, 2019-07-08:
When you run all unit tests, `pybamm.settings.debug_mode` is set to `True` and then the shape of every symbol you create is tested in `Symbol.__init__`. By default (running individual tests) `pybamm.settings.debug_mode` is `False`. Can you try setting `pybamm.settings.debug_mode = True` in the individual test?

martinjrobins, 2019-07-08:
not sure, the quick option should just run the unit tests as normal

tinosulzer, 2019-07-08:
So unit uses an executable whereas quick uses a subprocess. The executable ignores the debug mode option. Maybe easiest to just make unit also use a subprocess, and get rid of executables since we don't need python2 any more.
We should probably also set `pybamm.settings.debug_mode = True` in all the individual unit tests as well to avoid this kind of confusion

PR #515 (2019-07-08): #514 update get_dofs
# Description

change `get_dofs` to take an array

Fixes #514 

unit tests pass, but `run-tests.py --quick` fails for some reason??

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ python run-tests.py --unit`
- [ ] The documentation builds: `$ cd docs` and th [... cut]
closed 2019-07-08 by tinosulzer