pybamm-team/PyBaMM #532: Workshop 31/7 - 1/8

martinjrobins, 2019-07-11:
Proposed timetable:

### 31st July

10 - 11: Introductions, pybamm overview (@martinjrobins)
11 - 12:30: [Running a model](https://github.com/pybamm-team/PyBaMM/blob/master/examples/notebooks/models/SPM.ipynb) (@martinjrobins)
12:30 - 13:30: Lunch
13:30 - 14:30: [Changing the settings](https://github.com/pybamm-team/PyBaMM/blob/master/examples/notebooks/change-settings.ipynb) (@martinjrobins)
14:30 - 15:00: Battery models implemented in Pybamm
15:00 - 16:00: [Comparing models](https://github.com/pybamm-team/PyBaMM/blob/master/examples/notebooks/models/lead-acid.ipynb)
16:00 - 17:00: [Discretisations](https://pybamm.readthedocs.io/en/latest/tutorials/add-spatial-method.html) and [Solvers](https://pybamm.readthedocs.io/en/latest/tutorials/add-solver.html) in Pybamm

### 1st August

9:30 - 10:30 [Adding new models](https://pybamm.readthedocs.io/en/latest/tutorials/add-model.html)
10:30 - 12:30: Practical exercise: implement and solve a simple model in Pybamm (what is the simplest relevant model that we can get them to implement?)
12:30 - 13:30: Lunch
13:30 - 16:00: Discussion and planning: What do **you** want to do with PyBaMM? How can this be done? Any additional features in pybamm required? (outcomes here could be new issues for workshop participants / pybamm developers to work on)

Scottmar93, 2019-07-11:
Is the idea that everyone arrives with pybamm installed? Is this optimistic?

Scottmar93, 2019-07-11:
Regarding simple relevant models, I think (from easiest to hardest) the following:
- A simple diffusion equation in 1D (Heat transport in a rod / lithium transport in a particle) (we can get them to try different boundary conditions on this)
- The single particle model (basically just two diffusion equations (basically just 2 diffusion equations coupled together) (we can just have the voltage given by the open circuit voltage for simplicity)
- The single particle model with temperature (add a a temperature equation to the previous model and play around with boundary conditions)

Each of these models builds upon the previous and I think introduces various different features of pybamm.

tinosulzer, 2019-07-11:
Happy to run any of those sessions, and agree with Scott's comment and suggestions - showing how easy it is to change BCs would be especially nice. We could start with an ODE model (e.g. robertson) if we want something even easier. 
Note that there is a reasonably big step between implementing a whole model in one file and understanding/adding to the submodel structure we have, so we need to think about how to introduce that

rtimms, 2019-07-11:
I agree too. I think SPM is a good way to introduce submodels. We could get people to write a really simple particle submodel which takes `param` and `domain` and then sets the variables/boundary conditions appropriately for linear diffusion, then get them to create the spm as scott suggests. I think doing this outside of our existing submodels would be best, as they contain a lot of extra variables, broadcasts etc.

rtimms, 2019-07-11:
Following on, I guess it could go something like (part 0 solve an ODE?),  part 1 could be write a model for linear diffusion in a particle just in a script (as in the SEI example), part 2 turn this into a model class which you can call up and solve, part 3 could be turn this into a submodel and then create a model which uses it to make SPM. then extra work for people who are keen is extend it by adding extra eqns.

martinjrobins, 2019-07-11:
This sounds great. Do you think this will take much longer than the time I've currently allocated? We could either continue with the practical exercise after lunch and shorten the discussion, or remove/shorten some of the content on day 1. I'm leaning towards the former and shortening the discussion.

Is anyone keen to turn this into an exercise sheet? And implement some solutions? I'm thinking that after the workshop we might want to turn these into a 4-part notebook on writing a model in pybamm (the existing create-model notebook would be part 1)

Re installation issues. I've allocated an extra 30 mins for the 1st practical session. Hopefully this will be enough, but depends on how tricky the installation issue is!

rtimms, 2019-07-12:
yeah I can start taking a look at this today. I've made a new branch `training-workshop-solutions` where I've made the script, and will start to turn it into a model/submodel. I'll add a tex file here then we can make an exercise sheet. Once the solutions are done maybe we can just delete parts of it and do a fill-in-the-blanks type thing to help speed things along? I suspect doing all of what is outlined above will take longer than 2 hours, depending on people's experience. I guess if some people just get the script up and running, and have a tinker with it, then that is a success. The more confident people can then get into the model/submodel stuff

rtimms, 2019-07-12:
I started putting together some stuff for this. So far I have:

- exercise 1: solve an ode model
- exercise 2: solve a simple pde model
- exercise 3: add parameters and extra outputs to pde model
- exercise 4: solve a "real life" problem (few more parameters from actual battery related problems)
- exercise 5: take the script from the previous exercise and turn in into a pybamm model

I think this is already a fair amount to do in a few hours, but happy to gain/lose stuff where you see fit. The exercise sheet is fairly empty. For ex1 I've put in a semi-filled script which people can copy paste and fill in. What are people's thoughts on this style of sheet?

martinjrobins, 2019-07-14:
Thanks @rtimms, I think the exercise sheet looks very good, getting them to "fill-in" a script is a good way to get everyone started.

jedgedrudd, 2019-07-18:
Hi All,

I'm finalising the invitation email. Is there anything the attendees should install or sign up for in advance?

martinjrobins, 2019-07-18:
Ideally the attendees should have git, python 3.6+, jupyter-notebook and pybamm (with the scikits.odes optional dependency) installed. See the [README.md](https://github.com/pybamm-team/PyBaMM/blob/master/README.md) in the pybamm repo for install instructions. For windows users we recommend using Windows Subsystem for Linux (https://docs.microsoft.com/en-us/windows/wsl/install-win10), and installing all the above within this environment. 

We can help on the day with any installation issues, but it would be good if most people have the above installed before the workshop.

martinjrobins, 2019-08-15:
well done everyone, I think the workshop went very well! :)

closed 2019-08-15 by martinjrobins