pybamm-team/PyBaMM #533: Simplify issues on branch #488

Scottmar93, 2019-07-11:
@martinjrobins I am getting simplify errors on branch issue-488-backward-couple-thermal-equations (#488). When simplify is used, a domain is added to one of the parameters: "T_ref" which later results in domain errors. 

I tried to look through simplify by I didn't manage to distill the problem into a simple example. However, the behaviour can be seen by running 'test_spme.py' on the branch. It also seems to happen after simplifying different parts of the model if you run `test_spm.py`. I have put a hack (resetting the `simp._simplified_symbols={}`) with a `pybamm.logging` message in `ode_solver.py` which gets around this issue so can either look at this now or when the rest of the branch if finished.

Do you have any ideas what is going on?

martinjrobins, 2019-07-12:
yea, simplify messes with the tree, so domains can start to conflict with each other. The Simplification class should clear the domains of all variables, but this wasn't happening. I've fixed this now and pushed the change, so it should work now. 

Note that I've left in your hack, so you'll need to remove it when your ready

Scottmar93, 2019-07-12:
Awesome thank you Martin.

closed 2019-07-12 by Scottmar93