pybamm-team/PyBaMM #543: Enhancements to improve coupling to other software

TomTranter, 2019-07-19:
**Summary**
As far as I'm aware PyBaMM does not support saving and loading data for the state of the battery. This would be useful for initializing the variables for transient simulations. 

 **Motivation**
I would like to couple solving a transient heat conduction problem using PyBaMM for 1D electrochemistry and heat sources and OpenPNM for 2D or 3D conduction and heat removal. I envisage setting the state of a bunch of 1D PyBaMM models, running them in parallel for a certain period of time and then updating the global temperature in OpenPNM and feeding this back in a loop over time to update temperature dependent variables. I would therefore need to initialize the PyBaMM solvers with the variables such as concentrations and run them completely for a small time period or run them in a piecemeal fashion with the ability to call other programs between timesteps. I anticipate the first method to be simpler to implement and parallelize but perhaps not as efficient.

tinosulzer, 2019-07-19:
Sounds like a very sensible approach. You're right that there is currently no support for loading and saving (we've been using the `pickle` module when necessary). 
I think @rtimms has a bit of experience trying to use a bunch of 1D models coupled with a 2D conduction problem in FEniCS, so might have some good advice

TomTranter, 2019-07-19:
After some playing about I got a script working that passes back and forth average particle and electrolyte concentrations as initial conditions and also a heat source and iterates but it is slow because all the entire model structure is being created at each time step. It would be better to create the model once and be able to have greater control over time-stepping. So pass model objects back and forth in partially time-stepped states updating local temperature as you go.

TomTranter, 2019-07-19:
Saying this I haven't profiled the code yet and not sure if the main time is spent  setting up and calling the solver which would have to be done each time in either situation

tinosulzer, 2019-07-19:
>  It would be better to create the model once and be able to have greater control over time-stepping.

Yep absolutely! Feel free to edit `ODESolver.solve` or `ODESolver.integrate` however you need to achieve this. 
For some basic profiling, setting `pybamm.set_logging_level("INFO")` should at least give some insight into what the solver is spending time on. There is a bit of work setting up (simplifying, making jacobian, etc) which definitely doesn't need to be repeated every time in this case

rtimms, 2019-07-20:
may not be relevant, but the scikits solvers have a step function for solving odes/daes in a stepwise fashion see [here](https://github.com/bmcage/odes/blob/master/ipython_examples/Planar%20Pendulum%20as%20DAE.ipynb). I believe this is the way sundials recommends restarting their solver using the output of the previous step as initial conditions. This may be the way to keep stepping the same model forward, as by this point you aren't making a new model at each step with new initial conditions; you are just taking the discretised algebra system and solving with a new `y0`

TomTranter, 2019-07-23:
The way forward would be to have two functions in the solver that both utilize set_up. One called solve_all which accepts a list of times and one that accepts a timestep (dt). The last solution would be saved on the solver object for initializing the stepping to the next solution and solutions are passed back with each call. solve_all would return all solutions in with all timesteps as currently happens with solve and step would return solution for one timestep. The onus is then on the user to amalgamate the timesteps together should they wish. Sundials has similar functionality so it would be a case of writing the right wrapper functions and pointers to the sundial solver object

tinosulzer, 2019-11-20:
@TomTranter does anything still need to be done on this specific issue?

TomTranter, 2019-11-28:
Hi @tinosulzer @rtimms @Scottmar93 it would be good to sit down and review this together. I plan to come to Oxford again before xmas

Scottmar93, 2019-11-28:
Yeah good idea, @ferranbrosa was planning coming at some point before christmas as well so maybe we can do another implementation day to discuss these things?

ferranbrosa, 2019-11-28:
I am coming on the 9th of December, but I can join another day via Skype if needed.

tinosulzer, 2019-11-29:
I can join via Skype or will be there on 19th and 20th December

tinosulzer, 2020-05-18:
@TomTranter can this be closed now?

TomTranter, 2020-05-18:
Yep

closed 2020-05-19 by tinosulzer