pybamm-team/PyBaMM #563: add parameters by material

rtimms, 2019-08-01:
add parameters by choosing materials e.g. `pybamm.ParameterValues(cathode=...,anode=...)` then have library which allows user to pick and choose different materials

ferranbrosa, 2019-08-05:
Related to that, I think it would be interesting if the .csv file for the parameters included the parameters that are functions (including the name of the function to use), so just doing ``` pybamm.ParameterValues("input/parameters/lithium-ion/LGM50_parameters.csv") ``` would provide all the parameters. Of course, this could be still consistent of the parameters by material. Then, we could have subdirectories for each parameter function, and store the scripts there.

rtimms, 2019-08-05:
yeah that would be a neat way of doing that

tinosulzer, 2019-08-21:
I'll start working on this. @ferranbrosa how is progress with the LG M50 parameters? Just want to make sure we don't get too many conflicts

ferranbrosa, 2019-08-21:
I got the new parameters up and running. It will require some tuning so I will wait to make the pull request until I have finished all the simulations. Basically, I created a new CSV file with the parameters and new function files for the parameter functions, so I can make sure that when I make the pull request it is consistent with what you have done.

When changing the parameters I had some ideas (some of them I already mentioned them in this issue):
1. **Name of the files/functions:** we should agree on the notation for the file names. Apart from labeling the material (NMC, graphite...) we should include an identification of the battery (or manufacturer or whatever) because composition will change from one to another. Also the source of information (e.g. paper where it comes from).
2. **Parameter functions:** as I said above, it would be nice that the parameter could be defined at once by doing `pybamm.ParameterValues("input/parameters/lithium-ion/LGM50_parameters.csv")` or similar. At the moment, the parameter functions need to be specified separately.
3. **Initial conditions:** maybe they should be defined differently (as parameters vs initial conditions). Except for the electrolyte base concentration (which I think it should go together with other electrolyte parameters, and which is used for scaling) the other ones I think are not used for scaling (at least they are not in your paper).
4. **Interpolation:** it would be nice to consider the interpolation functions (discussed in #600) and how to best integrate it together.

I am happy to have a Skype and discuss about that if  you want.

tinosulzer, 2019-08-30:
Ok these are good points.

1. I'm thinking the file structure could be 
```
input
|-- lithium-ion
    |-- electrolytes
    |-- anodes
         |-- NMC
         |-- graphite
              |-- README.md (information about where the parameters came from)
              |-- base_parameters.csv (with links to the functions in the same directory)
              |-- diffusivity_authorYear.py
              |-- ocp_authorYear.py
    |-- cathodes
```
2. Following on from 1, the user specifies the relevant strings and the path to that folder is used
```
parameters = {
    "chemistry": ...
    "electrolyte": ...
    "anode": ...
    "cathode": ...
}
ParameterValues(parameters)
```
3. Where would you specify initial concentrations instead? In the same place as typical current?
4. Agreed but I think this separates well as a different issue

ferranbrosa, 2019-09-01:
1. I like the structure. Something it is not clear to me: within the "graphite" or "NMC" folders there would be several sets of parameters depending on the paper where they come from or each paper would be a different folder?
2. Sounds good.
3. We could define it in the same place as the typical current, if that makes sense within the file structure, as they are highly dependent on the battery (not just the chemistry or geometry, but all together). Therefore, we could define a "default" for simulations so when people call default parameters the set is complete and ready to run. For example, we could take as default 1C discahrge at room temperature, so that already defines initial conditions for concentrations and temperature, and a typical current.
4. Yes.

tinosulzer, 2019-09-05:
1. Seems safest to have separate folders for each paper (and then name the folders e.g. `graphite_authorYear`). Good to see all the parameters for e.g. the anode in one place, and if they are based on papers they should never change so there won't be a need to keep updating lots of different files
3. Sounds good

tinosulzer, 2019-10-05:
Needs discussion:
- move to `pybamm/parameters` folders since these are now "standard" parameter sets rather than user inputs?
- how to deal with miscellaneous parameters (e.g. electrode geometry, Bruggeman)?

Scottmar93, 2019-10-07:
Just reading up on this discussion ahead of the meeting. One thing related to initial concentrations (point 3.) is that it is probably useful to be able to convert an initial SOC into initial concentrations so we should have that flexibility in how we implement this (and not hard code initial concentrations into a csv). 

Perhaps a  `Study` (or `Experiment`) class would give good flexibility for interfacing to things like the initial conditions, cycling conditions etc. We could attach methods such as `set_charged` / `set_discharged` to set the initial conditions to fully charged or fully discharge. The initial temperature, voltage limits, how degraded the cell already is (once we have degradation), and things like figuring out what a 1C rate is (this will be dependent what the voltage limits are) could be set here as well. I think this could also be used to make it easier to design cycling routines (i.e. we could store a list of instructions somewhere `["discharge at 1C", "stop at v=3.2", "constant current charge at 0.5C", "Stop at V=4", "constant voltage"]` etc). Basically, have this class take care of things which need to be set for a specific experiment on a cell but could all be changed in another experiment.

rtimms, 2019-10-07:
added some comments [here](https://github.com/pybamm-team/PyBaMM/pull/647#issuecomment-538887190)

PR #647 (2019-10-14): Issue 563 parameters by material
# Description

- [x] Reorganised parameters by material.
- [x] Specified path to functions in csv files
- [x] Allowed setting either current or C-rate
- [x] Updated docs on how to add new parameters

Fixes #563 

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)

# Key checklist:

- [x] No style issues: `$ flake8`
- [x] All t [... cut]
closed 2019-10-14 by tinosulzer