pybamm-team/PyBaMM #569: sequence of jupyter notebooks / other types of documentation for model creation

martinjrobins, 2019-08-01:
**Summary**
a sequence of online tutorials (notebooks or other) going from creating simple to more complicated models in pybamm.

 **Motivation**
Try to ease new users into the pybamm framework, focusing in particular on users interested in model development 

**Potential list of tutorials**
(a) create an ode model
(b) create a pde model
(c) create a model class
(d) create a new submodel

jedgedrudd, 2019-08-01:
(e) how to make a reduced order model and compare against the full model

rtimms, 2020-01-14:
FYI I've started to add some notebooks based on the exercises from the last training workshop. These will roughly be 
1. ODE model (introducing the general model structure and solving)
2. PDE model (introducing geometry, mesh, etc.)
3. More complicated PDE based on diffusion in a particle (linked to a physical example, including parameters etc.)
4. Comparing the PDE for the particle problem to the ODE you get in the fast diffusion limit (introducing comparing solutions, and to act as an example of comparing the "full model" (PDE) to a "reduced order model" (ODE))
5. A modified version of the existing SEI model notebook. This is more complicated and introduces non-dimensionlisation etc. so that people's models can fit in with the existing framework.

Then possibly notebook on turning a model in a class and creating submodel classes, but as @martinjrobins  suggests, this might be better as a screen capture. As we discussed the other day, the submodel structure might be a bit intimidating for new users, so I think at this point in time it would be better for the notebooks to focus on creating a model all in the same script/notebook.

tinosulzer, 2020-01-14:
Yeah, I agree that the focus should be on creating a model all in a single script. Maybe even including parameters values in that too (instead of using standard_parameters_lithium_ion, at least for the examples, maybe even for when we do a full SPM and/or DFN in a single script).

So once we have a full DFN class for example, anyone wanting to add some new physics can just make a copy of that class with the new physics. And when they have finished, we take care of merging it into the submodel structure for them and adding an option to call that submodel, which shouldn't be too bad with all the different parameters and variables already defined (just moving things around).

PR #877 (2020-03-11): Issue 569 create model notebooks
# Description
Added some more notebooks covering creating models, and comparing full and reduced-order models. Also fixes some typos in other notebooks

Fixes #569 #672 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [... cut]
closed 2020-03-11 by rtimms