pybamm-team/PyBaMM #571: processed spatial variables

tobykirk, 2019-08-01:
We would like to process a SpatialVariable (into a "ProcessedSpatialVariable") to extract the limits and construct a np.array of points between them using linspace (or otherwise). This would allow easy plotting over any domain of choice.

martinjrobins, 2019-08-01:
might be able to make `ProcessedVariable` automatically detect if the variable is a spatial one and use a default np.array of points the same as the mesh on that domain?

rtimms, 2020-05-26:
you can now do this (see PR #1020) e.g.
```
import pybamm
import numpy as np

model = pybamm.lithium_ion.SPM()

sim = pybamm.Simulation(model)
solution = sim.solve()
r_n = solution["r_n [m]"]
r_n(r=np.linspace(0, 1, 10))
```
to get 10 linearly spaced points in the negative particle

closed 2020-05-27 by tinosulzer