pybamm-team/PyBaMM #577: Split diffusion equation and make fluxes state variables

hennessymatt, 2019-08-01:
**Summary**
Rather than solving the diffusion equation c_t = div(D * grad(c)) for the concentration c, allow this equation to be split into a 2x2 system given by c_t + div(N) = 0 and N + D * grad(c) = 0 that can be solved for the concentration and the flux N.

 **Motivation**
This will enable models that go beyond Fickian diffusion to be coded in PyBaMM where the flux depends on more than just the concentration gradient.  For example, stress-assisted diffusion leads to a flux that depends on pressure/stress gradients and phase-field models lead to a flux with third-order space derivatives.

martinjrobins, 2019-08-01:
can this be done already by specifying `N` and `c` as variables and using the two algebraic equations given above?

tinosulzer, 2019-08-01:
For finite volumes there will be a problem because `N` needs to be discretised to a StateVector with size `size(c) + 1` so that it evaluates on cell edges rather than cell centers.
But this should just be a case of making a new `FluxVariable` type object and then having an if statement in `disc.set_variable_slices` to set different slices for flux variables.

Note that as long as you can define the flux explicitly in terms of (multiple) variables then the current implementation should work ... e.g `N = D(c,T) * grad(c) + k * grad(p)` would be fine (not sure if this is the form you're thinking of @hennessymatt ?). Third-order space derivatives would definitely break something though.

hennessymatt, 2019-08-05:
That's pretty much the form I was thinking of @tinosulzer for stress-driven diffusion.  Splitting the diffusion eqn might also come in handy when prescribing boundary conditions for extended models as well, since conditions on the flux simply become Dirichlet-style conditions, and these would stay the same regardless of the form of the flux.

tinosulzer, 2021-03-29:
Not something we are planning on doing at the moment. Stress-driven diffusion has now been implemented separately

closed 2021-03-29 by tinosulzer