pybamm-team/PyBaMM #579: Add active and inactive material volume fractions

ferranbrosa, 2019-08-02:
**Summary**
In the current definition of the parameters the microstructure is defined by the porosity. I assume this means that the rest of the volume fraction is assumed to be active material. It would be interesting to split the volume fractions into electrolyte (i.e. porosity), active material and inactive material.

 **Motivation**
In some cases the conductive filler in the electrode can make up to 10% of the volume so that has an impact on the amount of active material to consider in the model.

tinosulzer, 2019-08-02:
> I assume this means that the rest of the volume fraction is assumed to be active material

Yes, in the electrode submodels the electrode volume fraction is just defined as `1-eps` (e.g. `sigma_eff * (1 - eps) ** bruggeman`

tinosulzer, 2019-11-12:
Need this now, @ferranbrosa have you started on this?

ferranbrosa, 2019-11-12:
I haven't, sorry! Go ahead with it.

tinosulzer, 2019-11-12:
No problem.
What is the symbol for active material volume fraction?
Also, renaming everything from "porosity" to "electrolyte volume fraction" is a bit of a pain but can be done, do you think this is necessary? Otherwise, we have "porosity" which refers to electrolyte volume fraction, and "active material volume fraction" which refers to electrode volume fraction

ferranbrosa, 2019-11-12:
Whatever you prefer for the notation, I have seen both terms used.

PR #726 (2019-11-13): Issue 579 volume fractions
# Description

Add tortuosity submodels, and add active/inactive material volume fractions

Fixes #579 
Fixes #648 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] A [... cut]
closed 2019-11-13 by tinosulzer