pybamm-team/PyBaMM #586: Set solver time in dimensional units

ferranbrosa, 2019-08-12:
**Summary**
Given that all the parameters of the model can be set to be dimensional, it would be useful to be able to set the time in the solver (t_eval) in dimensional units. For example, something like
`solve(model, t_eval, "hours")` or `solve(model, t_eval, "seconds")`.

 **Motivation**
In many cases, when we run simulations to compare with experimental data, we want to run it up to a given time which is in dimensional units.

martinjrobins, 2019-08-15:
I'd imagine that this would depend on how the model defines dimensional time. But presumably we can have a function in each model that returns the characteristic timescale of each model (@Scottmar93, is this the `tau_discharge` parameter?)

i.e.

```python
T0 = model.get_timescale().evaluate();
solve(model, t_eval/T0); # t_eval is in dimensional time units, which are defined by the particular model
```

tinosulzer, 2019-08-20:
The discharge timescale is parameter dependent (in particular, current-dependent) so you would need to also pass `param` to `get_timescale`. Otherwise it would have to use `model.default_parameter_values`.

ferranbrosa, 2019-10-01:
I got tau from `tau = param.process_symbol(pybamm.standard_parameters_lithium_ion.tau_discharge)` so probably we do not need a function for that. Shall we close this issue?

closed 2019-10-23 by tinosulzer