pybamm-team/PyBaMM #590: Termination events not consistent

ferranbrosa, 2019-08-13:
**Describe the bug**
I was running simulations on the LG M50 and I noticed a strange behaviour related to the termination events (strange behaviour for this case described below as it might not be that relevant). Trying to isolate what the problem was I ran it with the DFN and SPMe examples (Moura's parameters) and noticed that the termination events depend on t_eval which should not happen. Also, the solution does not provide the time and value of the event (probably because of the interpolation to t_eval).

Probably the reason is that the three events (max cp, min cn and min voltage) happen very close together so a high time resolution is needed to distinguish them.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to '/examples/scripts/DFN.py' or '/examples/scripts/SPMe.py'
2. Run the model for different number of points in the t_eval linspace.
3. **DFN:** if N = 100 the termination event is 'Maximum positive particle surface concentration' while if N = 1E4 the termination event is 'Minimum voltage'.
4. **SPMe:** if N = 100 the termination event is 'Maximum positive particle surface concentration' while if N = 5000 the termination event is 'Minimum voltage'.

**Additional context**
When running the LG M50 parameters for the SPMe I noticed that the terminal voltage can fall way below the cut-off voltage (set to 2.5V while the simulation went down to 2V). More interestingly, in other cases , when changing the value of the cut-off voltage it stopped at different points (which weren't the cut-off voltage) and the termination event was 'Minimum negative particle surface concentration'.

rtimms, 2019-08-13:
currently pybamm tries to work out the termination event by evaluating all the events at `solution.t[-1]` and `solution.y[:,-1]` (see `get_termination_reason` in `base_solver`. Note that this the last val in solution isn't necessarily the time at which the event occurs (it may even be very far away if `t_eval` is coarse... As you say, since the events all happen close together it doesn't always get it right. This would probably be improved somewhat by getting the solvers to spit out the time and statevector at the exact time the solver failed (if that info is available for all the solvers?). It could even be added into the solution object so it can be read as `solution.t_event` etc

ferranbrosa, 2019-08-13:
That would be interesting. I have never used ODE solvers in Python, but the MATLAB ones output t and y at the time of the event as optional outputs. It would be interesting to pick them up as for some batteries the potential near end of discharge is really steep, so the way it is implemented now requires very small time steps to make sure that this region is well captured.

rtimms, 2019-08-13:
yeah that would be nice. the scipy solver lets you access the time and event which caused it to stop (see `t_events` [here](https://docs.scipy.org/doc/scipy/reference/generated/scipy.integrate.solve_ivp.html)), but I dont know about the scikits ones. You could save the `t_event` from the scipy solvers and then fall back on `get_termination_reason` for the other solvers if they don't provide the time at termination

martinjrobins, 2019-08-15:
the scikits solvers return the times that events are calculated at in `sol.roots.t` and `sol.roots.y`, see https://bmcage.github.io/odes/dev/ode.html and https://bmcage.github.io/odes/dev/dae.html. So looks like we can get the exact time of the events for all solvers.

ferranbrosa, 2019-08-15:
I can give it a go if you want.

martinjrobins, 2019-08-15:
Great, sounds good @ferranbrosa

PR #595 (2019-08-16): Issue 590 termination events
# Description

I have added t_event and y_event to the object Solution that pick up the time and solution when the event happens. I have taken the format from Scikits and adapted the Scipy solver to provide a similar output. Also, I have changed `get_termination_reason` so now it evaluates t_event and y_event instead of the final time.

Fixes #590 

## Type of change

Please delete options that are not relevant.

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or fe [... cut]
closed 2019-08-16 by martinjrobins