pybamm-team/PyBaMM #591: Averaged OCP in SPM/SPMe

ferranbrosa, 2019-08-14:
**Describe the bug**
I have noticed that the averaged OCP for an electrode that both the SPM and SPMe provide does not correspond to the averaged OCP if you manually define it (you can do it several ways). Also, in some cases, there are some bumps appearing in the voltage that might be related to this issue.

I don't know if this issue applies to DFN as well as the new variables I defined do not work in that model.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'examples/scripts/SPMe.py' or 'examples/scripts/SPM.py'
2. Paste the bits of code below into the right places.
3. Run the script and observe the discrepancy between the average OCP that comes out of the model and the ones manually defined.

Optional:

4. Set the new maximum and initial concentrations to

```
cspmax = 45000
csnmax = 30000

param["Initial concentration in negative electrode [mol.m-3]"] = 0.98*csnmax
param["Initial concentration in positive electrode [mol.m-3]"] = 0.05*cspmax
param["Maximum concentration in negative electrode [mol.m-3]"] = csnmax
param["Maximum concentration in positive electrode [mol.m-3]"] = cspmax
```
5. You will notice some bumps in the positive OCP (these bumps what bothered me initially as they were even larger with my parameters).

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Additional context**
Definition of the new variables:
```
model.variables.update(
    {
        "Ferran's positive ocp": pybamm.standard_parameters_lithium_ion.U_p_dimensional(
            pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_p_xav), broadcast_domain=["positive electrode"]),
            pybamm.thermal_parameters.T_ref
        ),
        "Ferran's av positive ocp": pybamm.x_average(pybamm.standard_parameters_lithium_ion.U_p_dimensional(
            pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_p_xav), broadcast_domain=["positive electrode"]),
            pybamm.thermal_parameters.T_ref
        )
        ),
        "Ferran's other av positive ocp": pybamm.standard_parameters_lithium_ion.U_p_dimensional(
            pybamm.x_average(pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_p_xav), broadcast_domain=["positive electrode"])),
            pybamm.thermal_parameters.T_ref
        ),
        "Ferran's another av positive ocp": pybamm.x_average(pybamm.standard_parameters_lithium_ion.U_p_dimensional(
            pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_p_xav), broadcast_domain=["positive electrode"]),
            pybamm.thermal_parameters.T_ref
        ) / pybamm.standard_parameters_lithium_ion.potential_scale
        ) * pybamm.standard_parameters_lithium_ion.potential_scale,
        "Ferran's negative ocp": pybamm.standard_parameters_lithium_ion.U_n_dimensional(
            pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_n_xav), broadcast_domain=["negative electrode"]),
            pybamm.thermal_parameters.T_ref
        ),
        "Ferran's av negative ocp": pybamm.x_average(pybamm.standard_parameters_lithium_ion.U_n_dimensional(
            pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_n_xav), broadcast_domain=["negative electrode"]),
            pybamm.thermal_parameters.T_ref
        )
        ),
        "Ferran's other av negative ocp": pybamm.standard_parameters_lithium_ion.U_n_dimensional(
            pybamm.x_average(pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_n_xav), broadcast_domain=["negative electrode"])),
            pybamm.thermal_parameters.T_ref
        ),
        "Ferran's another av negative ocp": pybamm.x_average(pybamm.standard_parameters_lithium_ion.U_n_dimensional(
            pybamm.PrimaryBroadcast( pybamm.surf(pybamm.standard_variables.c_s_n_xav), broadcast_domain=["negative electrode"]),
            pybamm.thermal_parameters.T_ref
        ) / pybamm.standard_parameters_lithium_ion.potential_scale
        ) * pybamm.standard_parameters_lithium_ion.potential_scale
    }
)
```
(blame Scott for the terrible variable names 😛)

Plots:
```
time = pybamm.ProcessedVariable(
    model.variables['Time [h]'], solution.t, solution.y, mesh=mesh
)
x_averaged_OCV_p = pybamm.ProcessedVariable(
    model.variables['X-averaged positive electrode open circuit potential [V]'], solution.t, solution.y, mesh=mesh
)
x_averaged_OCV_n = pybamm.ProcessedVariable(
    model.variables['X-averaged negative electrode open circuit potential [V]'], solution.t, solution.y, mesh=mesh
)
P1 = pybamm.ProcessedVariable(
    model.variables["Ferran's positive ocp"], solution.t, solution.y, mesh=mesh
)
P2 = pybamm.ProcessedVariable(
    model.variables["Ferran's av positive ocp"], solution.t, solution.y, mesh=mesh
)
P3 = pybamm.ProcessedVariable(
    model.variables["Ferran's other av positive ocp"], solution.t, solution.y, mesh=mesh
)
P4 = pybamm.ProcessedVariable(
    model.variables["Ferran's another av positive ocp"], solution.t, solution.y, mesh=mesh
)
N1 = pybamm.ProcessedVariable(
    model.variables["Ferran's negative ocp"], solution.t, solution.y, mesh=mesh
)
N2 = pybamm.ProcessedVariable(
    model.variables["Ferran's av negative ocp"], solution.t, solution.y, mesh=mesh
)
N3 = pybamm.ProcessedVariable(
    model.variables["Ferran's other av negative ocp"], solution.t, solution.y, mesh=mesh
)
N4 = pybamm.ProcessedVariable(
    model.variables["Ferran's another av negative ocp"], solution.t, solution.y, mesh=mesh
)

plt.figure(1)
plt.plot(time(solution.t),x_averaged_OCV_p(solution.t),label="SPMe OCV")
# plt.plot(time(solution.t),P1(solution.t,x=0),label="P1")  # P1 produces NaNs
plt.plot(time(solution.t),P2(solution.t),label="P2")
plt.plot(time(solution.t),P3(solution.t),label="P3")
plt.plot(time(solution.t),P4(solution.t),label="P4")
plt.xlabel('t [h]')
plt.ylabel('Positive OCP')
plt.legend()

plt.figure(2)
plt.plot(time(solution.t),x_averaged_OCV_n(solution.t),label="SPMe OCV")
# plt.plot(time(solution.t),N1(solution.t,x=0),label="N1")  # N1 produces NaNs
plt.plot(time(solution.t),N2(solution.t),label="N2")
plt.plot(time(solution.t),N3(solution.t),label="N3")
plt.plot(time(solution.t),N4(solution.t),label="N4")
plt.xlabel('t [h]')
plt.ylabel('Negative OCP')
plt.legend()

plt.figure(3)
plt.plot(time(solution.t),x_averaged_OCV_p(solution.t)-x_averaged_OCV_n(solution.t),label="SPMe OCV")
# plt.plot(time(solution.t),N1(solution.t,x=0),label="N1")  # N1 produces NaNs
plt.plot(time(solution.t),P2(solution.t)-N2(solution.t),label="N2")
plt.plot(time(solution.t),P3(solution.t)-N3(solution.t),label="N3")
plt.plot(time(solution.t),P4(solution.t)-N4(solution.t),label="N4")
plt.xlabel('t [h]')
plt.ylabel('OCP difference')
plt.legend()
```

ferranbrosa, 2019-08-14:
It seems that this issue is related to the OCV entropic change. I have set that term to be zero in the functions and all the issues related to the OCPs disappear.

tinosulzer, 2019-08-20:
Is the issue that OCV entropic change is defined based on a different OCP?
If so, it would be really neat to define entropic change simply as `OCP.diff(T)` rather than manually having to define a different function every time you want to change the OCP

ferranbrosa, 2019-08-20:
I think there are two issues here. The first one was due to the stretch introduced in the OCP and not in the entropic change. As you can see in the images below, adding the stretch to the entropic change it improves.

However, notice that there is still some disagreement between the SPMe output and the other averaged quantities (which take the reference temperature), which makes me think there might be a wrong temperature somewhere (maybe on my code). In the manually defined averages, I pass `pybamm.thermal_parameters.T_ref` as the temperature for the OCV.

[image]
[image]

tinosulzer, 2019-08-20:
ok, I don't completely understand but sounds like you know what you're looking for

tinosulzer, 2019-10-25:
Did you manage to fix this @ferranbrosa ?

ferranbrosa, 2019-10-28:
Not actually. For my simulations I just switched it off as I am doing isothermal simulations.

rtimms, 2019-11-04:
definitely something odd going on with entropic change... I'm doing some comparisons with comsol and there are some discrepancies with some of the outputs in the negative electrode in particular. setting dU/dT = 0 fixes the problem. I'm looking into this now...

ferranbrosa, 2019-11-04:
I guess the first test would be to check if the temperature that the dU/dT term reads coincides with the set temperature in the isothermal case, to rule out the problem being the temperature being slightly off what it should be, as in all the plots above I passed the temperature as `T_ref`.

rtimms, 2019-11-04:
yeah. this is just in running normal DFN thermal model, not inputting an external temperature. in isothermal the temperature should just be the (broadcasted) scalar `T_ref`, but I'll check this is actually the case

rtimms, 2019-11-04:
@ferranbrosa  the entropic change was being evaluated at the non-dimensional T instead of the dimensional T. Let me know if this fixes the problem you were having

PR #702 (2019-11-05): #591 fix entropic change bug
# Description
Fixes a bug where the dimensional open circuit potential was being evaluated using the non-dimensional temperature instead of the dimensional temperature.

Also removed the params such as "Typical diffusivity..." etc from the csv files if a function has been provided. The typical coefficient is hard coded into the function, so changing the value in the csv had no effect and could potentially be confusing for users. 

Fixes #591 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/m [... cut]
closed 2019-11-05 by rtimms