pybamm-team/PyBaMM #606: Example notebookshowing changing submodel

rtimms, 2019-09-05:
Add an example notebook showing how to load up a default model (e.g. SPM) and swap out a submodel, e.g. ODEs for particle concentration (fast diffusion) instead of Fickian. Also show how to build up your own model from submodels.

rtimms, 2019-09-09:
Would simply having an option `build=True` by default for any of the pre-defined models be an easy fix for this? By default the model builds, but you can load it with `build=False` then swap out a submodel e.g. `model.submodels["electrolyte diffusion"] = my_new_electrolyte_model`, and run `model.build_model`? Thoughts @martinjrobins @Scottmar93 @tinosulzer

martinjrobins, 2019-09-09:
that sounds like a good solution to me

tinosulzer, 2019-09-10:
yep sounds good

PR #643 (2019-10-15): Issue 606 change submodel notebook
# Description
Adds a notebook which shows users how to swap out a submodel in an existing model, and how to build a new model using existing submodels. I've added an extra option `build=True` to the models so that you can load up a model without building (e.g. `pybamm.lithium_ion.SPM(build=False)`) then change a submodel before running `model.build_model()`. 

While doing this I though users might find it annoying that they have to be careful about which order they set the submodels in (for getting the coupled variables). I've changed the way `base_battery_mo [... cut]
closed 2019-10-15 by rtimms