pybamm-team/PyBaMM #619: Make pybamm pip installable

rtimms, 2019-09-16:


tinosulzer, 2019-10-14:
It seems casadi manage to wrap SUNDIALS and still pip install very very quickly. I have absolutely no idea how they do this but might be worth looking into @tlestang

tlestang, 2019-10-28:
@tinosulzer From what I've understood the whole of casadi is C++ so there is no need for wrapping the SUNDIALS.
Casadi just provides a python interface to the whole thing using [swig](http://www.swig.org/).

tlestang, 2019-10-28:
PyBaMM is now pip installable from the PyPi test index ([TestPyPi](https://test.pypi.org/))
```
pip install -i https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple pybamm
```
where `-i` is used to specify the test index as the main index and `--extra-index-url` allows the dependencies to be fetched from the main PyPi package index.

tinosulzer, 2019-10-28:
Great! That command worked for me :) Would be amazing if you manage to get SUNDIALS and KLU working automatically as well!

tlestang, 2019-11-04:
Add new functionalities to `setup.py` to automatically install SUNDIALS and scikits.odes.
Having downloaded the SUNDIALS source, running
```
python setup.py install
```
will install PyBaMM, build the Sundials and install scikits.odes.

The `setup.py` script now defines two custom commands:
```
python setup.py build_sundials --sundials-src=</path/to/src> --sundials-inst=<tgt/install/dir/> 
```
will build the sundials library from sources located in `path/to/src` into `tgt/install/dir`.
Additionally, the `build_sundials` command modifies the `activate` script of the current virtual environment to add the sundials directory to the `LD_LIBRARY_PATH`. If not in a python virtual environment, the user's `.bashrc` is modified instead.
```
python setup.py install_odes --sundial-inst=<tgt/install/dir>
```
installs scikits.odes using pip. If the `--sundials-inst` option was set to a nondefault location when building the sundials, then this location must be specified here as well.
Finally,
```
python setup.py install --sundials-src=</path/to/src> --sundials-inst=<tgt/install/dir/>
```
installs PyBaMM and runs both `build_sundials` and `install_odes` with the specified options.
```
python setup.py install --no-sundials
```
just install PyBaMM, without installing the sundials not scikits.odes.

tinosulzer, 2019-11-20:
Looks great @tlestang - can this be merged into master?

closed 2020-02-06 by tlestang