pybamm-team/PyBaMM #620: Simplifying jac slow for large systems

rtimms, 2019-09-16:
For large systems, the expression tree representing the Jacobian is very large, and simplifying it symbolically can be slow. Question: does `to_python` do some simplifications (e.g. multiply by zero)? Do some testing to see when to use simplifications

martinjrobins, 2019-09-18:
`to_python` will evaluate subtrees of the expression tree if they are constant, and thus simplify that way. But that is the only simplification.

At the moment the simplify is done in a bottom-up fashion, which is very slow. Performance would improve a lot if there was an initial simplify sweep top-down, for example pruning the tree when it finds a multiply by zero.

rtimms, 2020-07-05:
closing this as it is no longer slow if you use casadi (which is the default)

closed 2020-07-05 by rtimms