pybamm-team/PyBaMM #621: issues with jacobian if not using simplify

rtimms, 2019-09-17:
When running the lithium-ion battery models with `use_simplify = False`, one of two errors occurs:

1. For SPM and SPMe, the Jacobian can be calculated, but not converted to python (full error message below)

2. For DFN, the Jacobian cannot be calculated due to a domain error (full error message below)

To reproduce, run either the SPMe or DFN example scripts, setting `model.use_simplify=False`

1.
```2019-09-17 10:53:24,020 - [INFO] ode_solver.set_up(153): Converting jacobian to python
Traceback (most recent call last):
  File "examples/scripts/SPMe.py", line 28, in <module>
    solution = model.default_solver.solve(model, t_eval)
  File "/home/timms/Documents/PyBaMM/pybamm/solvers/ode_solver.py", line 37, in solve
    concatenated_rhs, y0, model_events, jac_rhs = self.set_up(model)
  File "/home/timms/Documents/PyBaMM/pybamm/solvers/ode_solver.py", line 154, in set_up
    jac_rhs = pybamm.EvaluatorPython(jac_rhs)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/evaluate.py", line 242, in __init__
    constants, self._variable_function = pybamm.to_python(symbol, debug=False)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/evaluate.py", line 203, in to_python
    find_symbols(symbol, constant_values, variable_symbols)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/evaluate.py", line 60, in find_symbols
    find_symbols(child, constant_symbols, variable_symbols)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/evaluate.py", line 60, in find_symbols
    find_symbols(child, constant_symbols, variable_symbols)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/evaluate.py", line 60, in find_symbols
    find_symbols(child, constant_symbols, variable_symbols)
  [Previous line repeated 2 more times]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/evaluate.py", line 55, in find_symbols
    constant_symbols[symbol.id] = symbol.evaluate()
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 157, in evaluate
    right = self.right.evaluate(t, y)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 156, in evaluate
    left = self.left.evaluate(t, y)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 90, in evaluate
    return self._concatenation_evaluate(children_eval)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 74, in _concatenation_evaluate
    return self.concatenation_function(children_eval)
  File "/home/timms/Documents/PyBaMM/venv/lib/python3.6/site-packages/scipy/sparse/construct.py", line 499, in vstack
    return bmat([[b] for b in blocks], format=format, dtype=dtype)
  File "/home/timms/Documents/PyBaMM/venv/lib/python3.6/site-packages/scipy/sparse/construct.py", line 555, in bmat
    A = _compressed_sparse_stack(blocks[:,0], 0)
  File "/home/timms/Documents/PyBaMM/venv/lib/python3.6/site-packages/scipy/sparse/construct.py", line 418, in _compressed_sparse_stack
    raise ValueError('incompatible dimensions for axis %d' % other_axis)
ValueError: incompatible dimensions for axis 1
```

2. 
```
2019-09-17 10:54:17,347 - [INFO] dae_solver.set_up(191): Calculating jacobian
Traceback (most recent call last):
  File "examples/scripts/DFN.py", line 28, in <module>
    solution = model.default_solver.solve(model, t_eval)
  File "/home/timms/Documents/PyBaMM/pybamm/solvers/dae_solver.py", line 76, in solve
    model
  File "/home/timms/Documents/PyBaMM/pybamm/solvers/dae_solver.py", line 192, in set_up
    jac_rhs = concatenated_rhs.jac(y)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/symbol.py", line 431, in jac
    jac = self._jac(variable)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 157, in _jac
    return SparseStack(*[child.jac(variable) for child in children])
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 157, in <listcomp>
    return SparseStack(*[child.jac(variable) for child in children])
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/symbol.py", line 431, in jac
    jac = self._jac(variable)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 346, in _jac
    left, right = self.orphans
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/symbol.py", line 192, in orphans
    return tuple([child.new_copy() for child in self.children])
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/symbol.py", line 192, in <listcomp>
    return tuple([child.new_copy() for child in self.children])
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 135, in new_copy
    new_right = self.right.new_copy()
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 135, in new_copy
    new_right = self.right.new_copy()
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 135, in new_copy
    new_right = self.right.new_copy()
  [Previous line repeated 3 more times]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in new_copy
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in <listcomp>
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in new_copy
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in <listcomp>
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in new_copy
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in <listcomp>
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in new_copy
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in <listcomp>
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in new_copy
    new_children = [child.new_copy() for child in self.children]
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/concatenations.py", line 94, in <listcomp>
    new_children = [child.new_copy() for child in self.children]
```
A whole bunch of `new_copy` stuff here from the list..., then 
```
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 335, in __init__
    super().__init__("*", left, right)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 88, in __init__
    left.auxiliary_domains, right.auxiliary_domains
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 127, in get_children_auxiliary_domains
    aux_domains[level] = self.get_children_domains(ldomain, rdomain)
  File "/home/timms/Documents/PyBaMM/pybamm/expression_tree/binary_operators.py", line 117, in get_children_domains
    ldomain, rdomain
pybamm.expression_tree.exceptions.DomainError: 
                children must have same (or empty) domains, but left.domain is '['current collector']'
                and right.domain is '['negative electrode']'

```

martinjrobins, 2019-09-18:
@rtimms: for the SPMe model, the problem comes when trying to evaluate this expression
```
+
├── 0.0
└── /
    ├── sparse stack
    │   ├── Index[0:40]
    │   │   └── *
    │   │       ├── -
    │   │       │   ├── 1.0
    │   │       │   └── Cation transference number = 0.4
    │   │       └── kronecker product
    │   │           ├── /
    │   │           │   ├── kronecker product
    │   │           │   │   ├── 0.0
    │   │           │   │   └── Column vector of length 1
    │   │           │   └── /
    │   │           │       ├── Negative electrode thickness [m] = 0.0001
    │   │           │       └── +
    │   │           │           ├── +
    │   │           │           │   ├── Negative electrode thickness [m] = 0.0001
    │   │           │           │   └── Separator thickness [m] = 2.5e-05
    │   │           │           └── Positive electrode thickness [m] = 0.0001
    │   │           └── Column vector of length 40
    │   ├── Index[0:25]
    │   │   └── *
    │   │       ├── 0.0
    │   │       └── Sparse Matrix (25, 120)
    │   └── Index[0:35]
    │       └── *
    │           ├── -
    │           │   ├── 1.0
    │           │   └── Cation transference number = 0.4
    │           └── kronecker product
    │               ├── /
    │               │   ├── -
    │               │   │   └── kronecker product
    │               │   │       ├── 0.0
    │               │   │       └── Column vector of length 1
    │               │   └── /
    │               │       ├── Positive electrode thickness [m] = 0.0001
    │               │       └── +
    │               │           ├── +
    │               │           │   ├── Negative electrode thickness [m] = 0.0001
    │               │           │   └── Separator thickness [m] = 2.5e-05
    │               │           └── Positive electrode thickness [m] = 0.0001
    │               └── Column vector of length 35
    └── /
        ├── Typical electrolyte concentration [mol.m-3] = 1000.0
        └── Maximum concentration in negative electrode [mol.m-3] = 24983.261993843702
```

For the sparse stack the middle branch results in a matrix of size 25x120, the other branches are vectors, which is why the stack fails. The simplify gets around this problem because the matrix in the middle is actually multiplied by a zero, so it gets simplified. But this seems to be a problem with the discretisation returning incompatible matrix dimensions for this model. Can you see why this would be the case?

martinjrobins, 2019-09-18:
for the DFN, seems to be an issue that the auxiliary domains in the provided model are incompatible. The simplify removes all domains, assuming they are correct, so this bug is not picked up if simplify is applied. But the problem is in the model construction or discretisation

rtimms, 2019-09-18:
thanks for investigating. for the DFN it seems that one of the operations which shifts an auxiliary domain to a primary domain isn't working as expected. one variable has primary domain ["negative electrode"] and then auxiliary domains ["negative electrode", "current collector"], but I think auxiliary domains should just be ["current collector"] (i.e. neg electrode has been promoted to primary domain due to some operation e.g. integration)

I'll take a look at this this afternoon

PR #628 (2019-09-24): Issue 621 jac no simp bug
# Description
Fixes a bug where when taking the jacobian of `Outer` when the left child evaluated to a number resulted in the wrong shape (this was the problem with SPM(e) when trying use jac without simplify).

`new_copy` now keeps the auxiliary domains as well as the domain (this was causing the problem when taking jac of DFN without using simplify). `jac` also now removes the auxiliary domains as well as the domain

Fixes #621 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist [... cut]
closed 2019-09-24 by rtimms