pybamm-team/PyBaMM #623: SOC calculation for Li-ion models

TomTranter, 2019-09-18:
**Summary**
The SOC is an important parameter and can also feature in calculating other constitutive relations such as dU/dT. A function is required which could even be used to initialize concentrations given the cell dimensions and capacity

**Additional context**
 The issue will also require implementing an r_average unary_operator. This is apparently tricky to do in a general way for SPM(e) and DFN

TomTranter, 2019-09-18:
```
rad = pybamm.SpatialVariable("r", [self.domain.lower() + " particle"])
c_s_r_av = pybamm.Integral(pybamm.x_average(c_s), rad)
```
works

TomTranter, 2019-09-24:
@tinosulzer @ferranbrosa @Scottmar93 @rtimms @martinjrobins : question and discussion on the SOC calculation. Assuming that the anode capacity is essentially limitless compared with the cathode the SOC of the battery is determined by the maximum and minimum level of lithiation in the cathode. A typical plot can be seen here for NMC 811 in figure 2 https://pubs.acs.org/doi/pdf/10.1021/acs.chemmater.9b00140
which has x-axis shown in Capacity and also "x" in Li_(1-x)Ni_(0.8)Mn_(0.1)Co_(0.1)O_2 where "x" is the extent of lithiation.
The theoretical minimum and maximum extent of lithiation fraction is between 0 and 1 but there are practical limits determined by setting an allowable voltage range which means that the SOC is usually measured and normalized over a restricted range. So should we actually be specifying the chemistry and initial average extent of lithiation and allowable range perhaps as the initial conditions as opposed to the concentrations. Setting the maximum concentration is a little trickier in some senses because how do you define your volume over which you are taking the average. Is it a pure crystalline primary particle? Is it an aglomerate secondary particles with interstitial space? Is it a volume averaged number including a battery level porosity. The x is in some ways easier to deal with and also less ambiguous in my view. What are your thoughts? This is also kind of a crucial concept for how to define SOC and also how to interpret experimental data that has been collected vs. SOC such as the entropic data

TomTranter, 2019-09-24:
On this point the key advantage of the 811 chemistry over 111 is the stability for a lower extent of lithiation which increases energy density as more of the theroetical maximum can be practically cycled so it's a very important material characteristic

ferranbrosa, 2019-09-24:
My understanding of the SOC is the following. I would like to discuss two main points: what do we understand as SOC and the range in which is defined. 

First, we should distinguish between the SOC of the battery and the SOC of the electrodes. The SOC of the electrodes (i.e. how much lithiated they are) is the one we need to evaluate the OCP or the dU/dT, while the SOC of the battery seems to be more an application indicator and not of much interest from the modelling point of view. However, notice that both the OCP and the dU/dT depend on the concentration at the interface, therefore it is not actually an SOC because the value depends on the dynamics and it can't be obtained just from Coulomb counting. For example, when you switch the battery off and observe relaxation, that is due to a variation in the OCP, but the SOC has not changed as there is no current applied. Therefore, we should be careful about what we define as SOC and decide if we refer to how much lithium is there in the electrode or the surface concentration (which could be normalised between 0 and 1).

About the range in which the SOC is defined, the experimentalists told me that they define SOC 0 and SOC 1 from the potentials. For example, for the cathode, they define SOC 1 to be 4.2 V and SOC 0 to be 3V (just made up the example so the values are most likely wrong). It does not mean that SOC 0 implies that there is no lithium in the electrode, as for some chemistries you must leave a certain amount of lithium in the electrode, otherwise, the structure is not stable and collapses. However, we can shift the origin of concentration and the equations remain invariant. Butler-Volmer changes, but given that the exchange current must be zero at SOC 0 and SOC 1, the definition is consistent. Therefore we can think of the lithium concentration in the electrodes as "active lithium" (or whatever name you prefer, not sure people have agreed on that) and we can neglect the fraction of lithium that remains in the electrode. 

Let me know if it makes sense. I am not sure if all this is obvious and well understood in the modelling community and I am just reinventing the wheel, or if there is some clarification to be made and which could lead to some model review paper. In any case, I am happy to have further discussions about the topic.

TomTranter, 2019-09-24:
Hi @ferranbrosa you make some very nice points. The relaxation especially is interesting from a modelling point of view too and I guess can be thought of as a redistribution of quickly accessible charge even though the average value is not changing. My understanding on the conventions in the modelling and experimental community about these definitions is certainly not great. And from a model validation point of view seems quite important. It's not a sexy thing to be working on (or reviewing) but quite useful in my view. Perhaps actually Greg and Emma have some input on this so we could formulate an email for them to verify

rtimms, 2019-09-24:
> while the SOC of the battery seems to be more an application indicator and not of much interest from the modelling point of view

> My understanding on the conventions in the modelling and experimental community about these definitions is certainly not great. And from a model validation point of view seems quite important.

I agree with both these points. Definitions of SOC in different modelling papers seem to vary (particular whether it is just defined as `c/c_max` or normalized. Would be nice to know if the experimental/parameteristaion people in Faraday have a standard for this (and other parameter names/definitions), particularly if we are wanting to link in to an experimental database at some point.

closed 2019-10-07 by TomTranter