pybamm-team/PyBaMM #646: Bug on Mac install (ft. sinister error message)

tinosulzer, 2019-10-04:
After running `pip install -e .[dev,docs]` and `python3 run-tests.py --unit`, `test_symobl_visualise` fails (hopefully just needs an extra dependency). 
NB - I don't know who thought it would be a good idea to call a function `_execute_child`
```
======================================================================
ERROR: test_symbol_visualise (test_expression_tree.test_symbol.TestSymbol)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/vsulzer/Documents/PyBaMM/tests/unit/test_expression_tree/test_symbol.py", line 312, in test_symbol_visualise
    rhs.visualise("StefanMaxwell_test.png")
  File "/Users/vsulzer/Documents/PyBaMM/pybamm/expression_tree/symbol.py", line 224, in visualise
    ).to_picture(filename)
  File "/Users/vsulzer/Documents/PyBaMM/PyBaMM-env/lib/python3.7/site-packages/anytree/exporter/dotexporter.py", line 268, in to_picture
    check_call(cmd)
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 342, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'dot': 'dot'
```

martinjrobins, 2019-10-05:
looks like `graphviz`  needs to be added as a dependency for PyBaMM. This provides the `dot` executable. Note this isn't pip installable, so user would have to install manually (https://graphviz.gitlab.io/download/)

ferranbrosa, 2019-10-06:
I noticed that when I ran the tests in my computer one failed (which was related to this `dot` thing).

PR #680 (2020-02-12): Issue 646 install
# Description

Update Linux instructions with @jedgedrudd 's comments and add instructions for MacOS

Fixes #646 
Fixes #673 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ]  [... cut]
closed 2020-02-12 by tinosulzer