pybamm-team/PyBaMM #651: Jacobian by equation

Scottmar93, 2019-10-09:
**Summary**
Perform the Jacobian calculation on each individual equation and then concatenate the result. 

**Reason**
Currently, we discretize equations and then combine into one large concatenated vector. This leads to very large expression tree which are lead to issues with RAM and are hard to interpret. We can instead simply hold off concatenating the discretized equations (i.e. move this step to the solver) and take the Jacobian of each equation individually. This helps with a few things: 
- it might help limit some of the RAM issues we were having on the larger expression trees
- it keeps the expression trees reasonably small and manageable
- we can add a "by equation" progress bar for calculating the Jacobian to help us understand what equations are slow
- we could calculate the Jacobian of different equations in parallel using pythons multiprocessing

rtimms, 2019-10-09:
something that might help is checking if any state vectors are in the tree in the base method for jac  [here](https://github.com/pybamm-team/PyBaMM/blob/41cb9dc7d5e3fe262067cd77eaa9c9233500ae03/pybamm/expression_tree/symbol.py#L429) and returning zeros of the right shape if not. not sure how often it happens but it might avoid calculating jac down a particular branch unnecessarily. should at least stop the tree for jac becoming so large

tinosulzer, 2019-10-25:
Can this now be closed @rtimms ?

rtimms, 2019-10-28:
yep. the functionality is there to do this if calculating the jacobian using the method in `discretisation` rather than during solver set up

closed 2019-10-28 by rtimms