pybamm-team/PyBaMM #662: Use fuzzy string matching

tinosulzer, 2019-10-14:
Fuzzy string matching libraries [already exist](https://chairnerd.seatgeek.com/fuzzywuzzy-fuzzy-string-matching-in-python/).
We could help the problem of "lots of variables in the variables dict" by implementing `model.variables` as a "fuzzy dictionary" which, when a key is required, does something like:

- try finding the key
- if `KeyError`, look in dict for similar keys (e.g. at most 3 most similar keys above a certain threshold of similarity)
- allow the user to change the key they're asking for
- continue

Scottmar93, 2019-10-16:
Seems a cool idea. Would be good to return a warning that says: 'couldn't find "x" so using "y" instead'. I think we should also have the option to turn on/off this fuzzy key so that we can raise error when running tests.

tinosulzer, 2019-10-17:
Oh yeah I hadn't thought of the tests. Maybe easier to just raise an error that suggests a different name instead (the same way git does when you type in the wrong command)

Scottmar93, 2019-10-22:
Do we not already have that type of functionality though? In that if we type the wrong string in we will see an error message saying something along the lines of "domain must be in ["negative particle", ... etc], and not X"? It won't suggest the closest matching string to your domain but given there is only about 5 domains so its not hard for the user to see what they should have typed.

PR #796 (2020-01-28): Issue 790 param not recognized
# Description

Add fuzzy dict and raise errors if a parameter is updated that wasn't previously defined (so users don't think they've updated them when they haven't)

Fixes #790 
Fixes #662 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that sp [... cut]
closed 2020-01-28 by tinosulzer