pybamm-team/PyBaMM #671: Add mechanical submodel for particles

Scottmar93, 2019-10-21:
**Summary**
Add a new submodel for the particles which accounts for mechanical stress and stress assisted diffusion. 

**Additional context**
I think these are the key steps for implementing this issue. 

1. First write a test that outlines how you would like the new submodel function. I think that loading the model so that you can run it will likely look something along the lines of:
```
options = {"Mechanics": True}
model = pybamm.DFN(options)
``` 

2. Add any new dimensional parameters to the input parameters file somewhere appropriate. 

3. To fit in with the rest of the PyBaMM structure the model will first have the be non-dimensionalised i.e. appropriate scalings such as ```c_dimensional = c_typical * c_dimensionless``` must be chosen and then plugged into the model equations. 

4. Add any new dimensionless parameters to the `standard_parameters.py` files

5. Write new governing equations, boundary conditions, and initial conditions in a new particle submodel following the structure of the `fickian` submodels folder. (you should write the new governing equations in both many particle and single particle format so that it will work on the DFN, SPM, and SPMe as well as the 3D models)

tinosulzer, 2019-10-21:
If possible, we should try to rewrite the particle submodels so that you don't need to write the governing equations out twice for both many particles and single particles. Applies to this issue and other extensions (e.g. distribution of particle sizes)

aishine, 2019-12-03:
Hi all, an update for this issue. I have added stress-coupled diffusion function to the file "base_fickian_particle_mechanics.py" and a notebook of "Running mechanical-coupled model in PyBaMM.ipynb". It is a draft and not tested. Below are some problems I need help:

1. How to add a set of parameters for a battery? I have a benchmark for the Enertech LiCoO2-Graphite cell, and I think it will be good to add this to PyBaMM and for validation of the results.
2. How to add outputs to PyBaMM results? 
3. How to call the specific submodel above? I found one example for the SPM, but it is not straight to DFN model.

tinosulzer, 2019-12-05:
Hi @aishine , looks like a good start. To answer your questions:

1. There is information about how to do this [here](https://pybamm.readthedocs.io/en/latest/tutorials/add-parameter-values.html) -  let me know if you have any questions
2. Outputs are in the `variables` dictionary which are updated individually by each submodel in `get_coupled_variables` (e.g. [here](https://github.com/pybamm-team/PyBaMM/blob/d8d24c60ecf83800255b50664626f1aba3919606/pybamm/models/submodels/particle/fickian/fickian_single_particle.py#L40) for Fickian diffusion for SPM). This can be a little bit confusing as we use subclasses to avoid duplicate code, so you have to follow the trail of parent classes to find specific functions. The easiest thing to do at this stage is just to add the variables to the dictionary [here](https://github.com/pybamm-team/PyBaMM/blob/d8d24c60ecf83800255b50664626f1aba3919606/pybamm/models/submodels/particle/base_particle.py#L60) and we can think about how to do it better later 
3. Don't worry about this for now - actually I would just start by hardcoding mechanics into `base_fickian_particle` and then add the option to change it later

To add new parameters:
```python3
nu_p = pybamm.Parameter("Poisson's ratio for cathode")
E_p = pybamm.Parameter("Young's modulus for cathode [GPa]")
c_p_0 = pybamm.Parameter(
    "Reference concentration when cathode particle is free of stress"
)
Omega_p = pybamm.Parameter("Partial molar volume for cathode, Dimentionless ")
theta_p = (
    Omega_p ** 2 / R * 2 / 9 * E_p * (1 - nu_p)
)  # intermediate variable  [K*m^3/mol]
```
You can either hardcode this into `_flux_law`, or add to `parameters/standard_parameters_lithium_ion.py` and then call only `theta_p` using `param.theta_p`. Then you also need to add "Poisson's ratio for the cathode", etc to the csv file of parameters

weilongai, 2020-07-07:
Sorry for not updating this branch for a while. I'm currently working on the model within an independent branch of PyBaMM, with the help of Diego @dalonsoa. When this work is finished, we will merge to this branch.

PR #1232 (2020-11-16): Cracking submodel
# Description
 A new submodel for particle cracking in pybamm.  No dependencies that are required for this change.

Fixes #671

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# [... cut]
closed 2020-11-16 by tinosulzer