pybamm-team/PyBaMM #688: Add simulation class

Scottmar93, 2019-10-26:
**Summary**
Create a simulation class which wraps model, discretization, parameters etc. 

**Motivation** 
Can be extended to allow linking of the experiment, model, discretization, and solver together and allow each to reference each other. Also simplifies the pybamm user interface so that users only need to see what the actually are setting

tinosulzer, 2019-10-26:
See #39 and the branch `issue-39-simulation-class` - happy for you to take over on this though :)

Scottmar93, 2019-10-28:
Ah nice thought you had deleted the branch so just made a quick mock up of what the class would look like. Will take a look at #39 as well.  Doesn't seem like there is too much to do but might be nice to have some reset type features that would allow you to undiscretize your model etc.

tinosulzer, 2019-10-28:
Sounds good, note you can do `disc.process_model(model, inplace=False)` to leave the original model un-discretised

PR #693 (2019-10-31): Issue 688 add simulation class
# Description

Convenience class for interacting with the front-end of PyBaMM. I have set it up to try to be Jupyter notebook friendly in that things won't break if you try to build twice or solve without first building. It'll also automatically reset everything if you decide to change a spatial_method / parameter etc. Would be good to add some functionality on saving an loading later as well as allowing for experiment specific things to be set such as the current. 

Fixes #688 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https [... cut]
closed 2019-10-31 by Scottmar93