pybamm-team/PyBaMM #695: Improve interface for post-processed variables, including exporting

tinosulzer, 2019-10-30:
**Summary**
 Add functionality to simulation class to post-process and export variables

 **Motivation**
Allow users to export variables to csv, matlab, etc for visualisation

TomTranter, 2019-11-04:
How would this handle the various spatial and temporal differences in the data? As there are many permutations might be better just to leave this in the users hands.

TomTranter, 2019-11-04:
After our conversation about this - I think it would be really useful to group the variables into domains so that they can all be visualized in the same way. If you run the following on a 1 plus 1D it fails on about half the variables
pybamm.post_process_variables(variables=spm.model.variables, t_sol=spm.solution.t, u_sol=spm.solution.y, mesh=spm.mesh)
maybe the post_process_variables and any new functions could accept a domain argument and only spit back the ones defined on that domain. Or if no domain is specified is could return a dict of dicts with the domain as the key on the first level

rtimms, 2019-11-04:
following on from that, it would be really great in Nplus1 to be able to visualize or create post process variables for the 1D model at a given y,z coordinate. But maybe this is more difficult than just improving how processed vars deals with higher-D variables...

tinosulzer, 2019-11-05:
ProcessedVariable is a hacky mess atm so I am very happy for you to completely overhaul it if you have clear ideas how to do it. There's no support for 4D objects (t+r+x+z) atm hence why 1 plus 1D SPM fails.

PR #811 (2020-02-11): Issue 695 export variables
# Description

Add some more ways to export variables (matlab and csv file), to then post-process using different tools.

Fixes #695 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fi [... cut]
closed 2020-02-11 by tinosulzer