pybamm-team/PyBaMM #712: helper function for changing BCs

rtimms, 2019-11-07:
It would be nice to be easily update boundary conditions after loading up a model.

It might also be easier to navigate the model boundary conditions if the keys of the dict were the variable name string rather than the object itself, as it is currently hard to quickly manually edit BCs. I', not sure if this causes problems elsewhere during discretisation though. In any case a helper function would be useful.

Scottmar93, 2019-11-07:
Helper function could take in a variable name string, look through the variables to see what the symbol is and then find the corresponding boundary condition. This avoids needing to change the structure of boundary conditions elsewhere

tinosulzer, 2021-03-29:
Isn't it just this?
```python3
c_e = model.variables["Electrolyte concentration"]
model.boundary_conditions[c_e] = {
    "left": (pybamm.Scalar(1), "Neumann"),
    "right": (pybamm.Scalar(-1), "Neumann"),
}
```
Not sure this warrants a helper function?

rtimms, 2021-03-30:
Agreed, closing.

closed 2021-03-30 by rtimms