pybamm-team/PyBaMM #718: alternative geometries for lumped thermal models

rtimms, 2019-11-08:
At the moment the lumped thermal models assume a pouch cell geometry. We could get users to specify the cell surface area and volume in the parameters file so that the correct ratio can be applied in the Newton cooling. This would be useful if you wanted to do a 1D electrochemical model, with a lumped thermal model for a specific cell geometry.  

The Nplus1D thermal models also assume a pouch cell model and will probably remain this way for now as they account for cooling from the face of the current collector (which would be would up in a jelly roll and therefore not open to the air for cooling). It should probably be clearer that these models are for pouch cells only.

ferranbrosa, 2019-11-11:
If I understand correctly this affects only the cooling area that multiplies the cooling coefficient. If that is the case, should we just let the user specify this area, so there is no need to consider the different cases separately (pouch, prismatic and cylindrical)?

rtimms, 2019-11-11:
yeah exactly, my picture was when the user puts in parameters, they explicitly add surface area (but surface area **for cooling**, as you suggest, is more precise) and volume for that cell. and then in the submodel we use those params so we don't care what the geometry actually was

ferranbrosa, 2019-11-11:
In fact, unless you need the volume of the cell for some other calculation, you could just pass the surface area for cooling per unit of volume (similar to the specific area defined for the electrode particles). Or maybe that can be confusing for the users?

rtimms, 2019-11-11:
yeah i'd be happy with either. might be useful to have volume for other calculations after the fact such as variable x per unit volume.

ferranbrosa, 2019-11-11:
Makes sense. Then I would define cell volume (which if undefined could be calculated from the electrode thickness and so on) and the cell cooling area so users can play with it.

ferranbrosa, 2020-04-21:
Following this discussion from a few months ago, it seems that the idea is to pass battery cooling surface area and battery volume as parameters (within the cell parameters, in the macroscale geometry section).

Would it make sense to define that if these two variables are not defined then it assumes a pouch cell? Or is it better to leave it to break and update all the parameter files?

rtimms, 2020-04-21:
Maybe we should be explicit in making people define the cooling area and volume. I think the less we assume the better, as people might easily miss assumptions that are hidden away.

rtimms, 2020-05-22:
Not sure what the latest is on this, but my thoughts for how to do this to best fit into the structure would be to pass an option `geometry` to the lumped model which can, for now, be "arbitrary" (default) or "pouch". Then just put an if statement in the either uses the surface area to volume ratio for cooling param or computes the cooling for the pouch (we could add other popular geometries if we wanted). 

Then was choosing the thermal model as an option to the battery model we could do "lumped", "lumped (pouch)", etc.

In the parameters csv we can just put a single value in for the urface area to volume ratio for cooling  which we leave people to compute, along with a comment saying what geometry that was computed for.

ferranbrosa, 2020-05-22:
Sounds good. Do we want to pass as a cell parameter the surface area to volume ratio or pass surface area AND volume?

rtimms, 2020-05-22:
not sure on that. probably pass both as the volume is fixed, but the cooling area can change (e.g. only cooling part of the cell)

PR #1022 (2020-05-28): Issue 718 cooling area
# Description

The lumped thermal model now takes an option to choose between arbitrary geometry (default) or a pouch geometry. The arbitrary geometry takes a total heat transfer coefficient, surface cooling area and cell volume. The pouch geometry calculates it from the different parts of the pouch (including tabs). 

Fixes #718

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the li [... cut]
closed 2020-05-28 by rtimms