pybamm-team/PyBaMM #720: Uniform grid sizing

tinosulzer, 2019-11-08:
Add functionality which allows the user to specify grid points in one subdomain and others get chosen automatically for a uniform grid size (in the x-dimension)

hemantsoni42, 2023-01-29:
Hello @tinosulzer , I am interested in contributing to the GSOC 2023 project and would love to work on one of the open issues. Can you please assign me this one so I can start contributing? 
Can you please guide me for this issue because this is my first contribution. And please let me know in which file this functionality will coded.

Thank you Sir

RohitP2005, 2024-12-17:
Is the issue still open , if yes i would like to solve

RohitP2005, 2024-12-22:
@valentinsulzer can u assign me this issue, i can work on
 this

RohitP2005, 2024-12-24:
hey @valentinsulzer 
If u don't mind I’d appreciate some direction on where to start. Are there any specific files or areas I should focus on first?

swastim01, 2025-10-30:
I noticed this issue had been open for some time, so I went ahead and implemented the functionality.
I’d appreciate any feedback or suggestions for improvement from the maintainers.