pybamm-team/PyBaMM #722: Print equations nicely and without expanding tree

TomTranter, 2019-11-11:
**Summary**
Rendering the expression tree is a giant mess and impossible to decipher. It would be great to be able to see what the equations are for each function/variable in one line without rendering the whole tree underneath. Also rendering them in Latex would be good. Sympy is able to render equations in Latex very nicely

TomTranter, 2019-11-11:
https://docs.sympy.org/latest/tutorial/printing.html

tinosulzer, 2021-01-25:
The main challenge here is with parameters that are combinations of other parameters: how do we know what to call them if we are not expanding them?

A nice workaround is to overload the `__setattr__` method in `LithiumIonParameters` (more  specifically, in a base `Parameters` class) as follows:
```python3
def __setattr__(self, name, value):
        if isinstance(value, pybamm.Symbol):
            value.name_for_printing = name
        super().__setattr__(name, value)
```

Then anywhere else in the class if we do something like
```python3
self.L_x = self.L_n + self.L_s + self.L_p
```
the rhs symbol (an `Addition`) will be given a print name of `L_x`. 

Similarly we can define print names for all kinds of symbols, and then we just need to convert to sympy for printing

tinosulzer, 2021-02-01:
https://github.com/connorferster/handcalcs

PR #1568 (2021-08-09): Add latexify
# Description

- Add latexify in expression_tree/operations and BaseModel
- Convert eps to epsilon in print_name.py
- Add some print_name

Fixes #722 
## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which [... cut]
closed 2021-08-09 by tinosulzer