pybamm-team/PyBaMM #740: Add presolve ability to models

Scottmar93, 2019-11-21:
For models like the SPMeCC and others we can decouple some of the equations and presolve them. It would be good if we could do this within a single model so that useful variables can still be easily accessed. 

**Possible approach**
_Model stage:_
1. Attach a `presolve=True/False` flag to each submodel, create `self.presolve_algebriac`, etc dictionaries. 
OR 
2. Search through the algebraic equations and if it is found that they do not depend any other variables in the problem and also have no time dependence then add that algebraic equation to the presolve list along with the corresponding boundary conditions and initial conditions (for the variable in that algebraic equation). 

*Discretization and solving:*
1. Upon discretization, indicate the components of y_slices which are to be presolved (possibly by putting these to the end of the state vector similar to how external variables are treated). In the solver stage, first presolve the variables which need to be presolved and then proceed in the same manner as how external variables have been treated.

tinosulzer, 2019-11-21:
This would be pretty cool - especially the second option. e.g. for lead-acid composite models, solve the leading-order model first and then use that for the composite model. You might be able to do it by looking for upper / lower triangular blocks in the jacobian.

tinosulzer, 2021-09-10:
Can this be closed? @rtimms @Scottmar93

rtimms, 2021-09-10:
Yeah, I think the way around this is to just have separate models and solve them sequentially, extracting the variables you need. I don't think it happens often enough/is worth the effort to automate this. 

This is kind of how the [pouch cell model](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/models/pouch-cell-model.ipynb) works. You solve a model for the effective resistance and one for the average through-cell behavior, then post-process to combine the results. I guess if one model depends on another (e.g. solving leading then first order) you could pass the leading-order variables in as external variables to the first-order model. Still, don't think it's worth automating/having as an option.

closed 2021-09-10 by tinosulzer