pybamm-team/PyBaMM #750: specs for pybamm jupyterhub machine

martinjrobins, 2019-11-27:
Here are my thoughts on what we need for a jupyterhub machine:

I can see two modes of operation, one with relatively low usage for individual people trying out pybamm in their own time, the other during workshops where we will have high usage. So I would think that a virtual machine that we can resize during a workshop is the best way to go.

One option would be to go with one of the cloud providers, for example azure, swapping between two vm types:

For regular use:
1 D2 v2 (2 vCPU(s), 7 GB RAM); Linux – Ubuntu; 3 year reserved; 1 managed OS disks – £38.35 per month
For workshops (we'd only pay for time we need here, rather than per month):
1 D14 v2 (16 vCPU(s), 112 GB RAM); Linux – Ubuntu; 3 year reserved; 1 managed OS disks – £315 per month

For the latter configuration we can even get people to indicate whether they want to use the jupyterhub machine for the workshop, and resize the machine appropriately. Note, however, that the maximum size in this class of vm is the D15 (20 cpus, 140GB of ram). If we would need more than this we should choose a different class

The other option is to host out own physical machine somewhere, but this would limit the maximum specs we could use for workshops to whatever is available. This might or might not be a problem, depending what is available for us to use?

tinosulzer, 2019-11-27:
Sounds like a good option if availability of physical machines is limited. As someone mentioned in the meeting, I would hope that, in the regular use case, it acts more as a "try before you buy (i.e. invest the ~0.5-2hrs to install locally for free)", rather than something people use regularly. How strongly we suggest / enforce (e.g. max one month remote use?) this is up for debate. Similarly at the end of the workshops after people have used it and are happy with it we could leave an hour open to help install locally for future use.
 
In the "regular use" case, can we monitor how much use it is getting, and then increase/decrease specs as necessary? And do we have to pay upfront for x months or is it a rolling monthly thing that we can try out?

martinjrobins, 2019-11-27:
you can certainly monitor how much use it gets, azure gives you all the analytic data you need. Re the pricing, the prices I quoted are for 3 years reserved instances, and you can pay monthly. You are free to change the specs of the machine up or down the particular class of machine you reserve, but you are basically locked into a 3 year contract. You can also do a 1 year reserved machine, or a pay-as-you-go, with the prices going up accordingly

jedgedrudd, 2019-12-02:
Amazon also offer a VM service: [https://aws.amazon.com/ec2/?c=7&pt=1](https://aws.amazon.com/ec2/?c=7&pt=1) and this can be an auto-scaling service: [https://aws.amazon.com/ec2/autoscaling/?c=7&pt=2](https://aws.amazon.com/ec2/autoscaling/?c=7&pt=2)

I am looking for a PostGreSQL server, and both Amazon and Azure offer these. Whichever we choose, it would make sense to pick the same one.

martinjrobins, 2019-12-03:
agree we should stick with a consistent cloud provider. I'm happy with either Amazon or Azure, with a slight preference for Azure only because I'm more familiar with their system.

TomTranter, 2019-12-04:
Could we just get some permanent resource on Michael?

tinosulzer, 2020-01-27:
I think we can close this?

martinjrobins, 2020-01-27:
yup

closed 2020-01-27 by martinjrobins