pybamm-team/PyBaMM #759: DAE solver for non-smooth currents

tinosulzer, 2019-12-11:
Fix (at least one of the) DAE solvers to work when the input current is not smooth.

Hacky approach is to do step-by-step and match up discontinuities with steps

martinjrobins, 2020-01-14:
could we augment `model.events` to take 2 types of events: "termination" and "discontinuity". The solver would then make sure to integrate up to just before a discontinuity, and restart integration just after the discontinuity.

tinosulzer, 2020-01-14:
Good idea, so would the discontinuities be a list of variables for which to look out for discontinuities? Or do we analyse the input current (a parameter) before solving to identify discontinuities and then tell the model/solver about them before solving?

martinjrobins, 2020-01-14:
I'm thinking that the user would manually specify where the discontinuities occur in time, based on the parameters of the model. Therefore discontinuities would be an expression that evaluates to the time when the discontinuity takes place. 

So, given a model that has parameter `a` and which exhibits a discontinuity at `2*a`, the user would set `model.events = [(pybamm.DISCONTINUITY, 2*a)]`

Then it would be easy for the solver to get the list of where discontinuities occur in time, and make sure that the integration is restarted at each discontinuous time-point

martinjrobins, 2020-02-03:
Each event now needs 3 pieces of info, its name, expression and type (DISCONTINUITY or TERMINATION). I think this needs a separate class to encapsulate this (and provide some documentation). Implemented this as `Event`, so that to create some events you might do something like this:

```python
        model.events = [
            pybamm.Event("var1 = 1.5", pybamm.min(var1 - 1.5)),
            pybamm.Event("var2 = 2.5", pybamm.min(var2 - 2.5)),
            pybamm.Event("nonsmooth rate",
                         pybamm.Scalar(2.5),
                         pybamm.EventType.DISCONTINUITY
                         )
        ]
```

PR #808 (2020-02-09): Issue 759 non smooth dae
# Description

Fixes #759 

- adds `pybamm.Event` class for holding information about events
- added a new class of event that returns the time that a discontinuity in the solution occurs
- changed the BaseSolver to filter out the discontinuity events. Integration is split over multiple segments of the time series, new time points are added just before and just after the discontinuity and added to the solution object
- for models using heavyside functions in certain patterns, the discretisation class automatically adds in discontinuity events required

## Type of chang [... cut]
closed 2020-02-09 by martinjrobins