pybamm-team/PyBaMM #784: Simplifying solver interface

tinosulzer, 2020-01-13:
It should be possible to merge 

- `base_solver`, `ode_solver` and `dae_solver` into a single base class
- `scikits_ode_solver` and `scikits_dae_solver` into a single class

Do you think this would simplify the interface @rtimms ?

rtimms, 2020-01-14:
yeah I think that would be good. could we also remove `compute_solution` and call `integrate` directly from `solve` and `step`? `compute_solution` saves rewriting the little integrate, get termination reason block of code in both `step` and `solve`, but it wouldn't be a massive bit of repetition and would probably be easier to follow for new users. (I think I introduced `compute_solution`, but I don't like it now haha)

I think the only solver with a drastically different interface is casadi, and we already overwrite some methods there anyway.

tinosulzer, 2020-01-14:
Ok, that sounds good - no worries, `compute_solution` seemed like a good idea at the time haha

PR #800 (2020-02-04): Issue 784 simplify solver
# Description

Simplify the various solver classes by merging the `set_up` methods and getting rid of `compute_solution`. Also combined the implementations for external variables and inputs

Fixes #784 
Fixes #789 
Fixes #463 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimi [... cut]
closed 2020-02-04 by tinosulzer