pybamm-team/PyBaMM #788: EvaluatorPython ValueError - shape of data problem

TomTranter, 2020-01-17:
**Describe the bug**
EvaluatorPython seems to handle data for multiple timesteps fine for some variables but not others

**To Reproduce**
Steps to reproduce the behaviour:
```
import pybamm
import numpy as np
import matplotlib.pyplot as plt
from pybamm import EvaluatorPython as ep
plt.close('all')

model_options = {
    "thermal": "x-lumped",
#    "external submodels": ["thermal"],
}
model = pybamm.lithium_ion.SPMe(model_options)
solver = pybamm.CasadiSolver()
#solver = model.default_solver
sim = pybamm.Simulation(model, solver=solver)
t_eval = np.linspace(0, 0.01, 10)
T_av = 0.0
for i in np.arange(1, len(t_eval) - 1):
    dt = t_eval[i + 1] - t_eval[i]
#    external_variables = {"X-averaged cell temperature": T_av}
    external_variables = None
    T_av += 1.0
    sim.step(dt, external_variables=external_variables)

eval_func_a = ep(sim.built_model.variables["Measured open circuit voltage [V]"])
eval_func_b = ep(sim.built_model.variables["X-averaged total heating [W.m-3]"])

t_last, y_last = sim.solution.t[-1], sim.solution.y[:, -1]
t_all, y_all = sim.solution.t, sim.solution.y

variables = {
    "Measured open circuit voltage [V]": sim.built_model.variables[
        "Measured open circuit voltage [V]"
    ],
    "X-averaged total heating [W.m-3]": sim.built_model.variables[
        "X-averaged total heating [W.m-3]"
    ]
}
var = pybamm.post_process_variables(
    variables, sim.solution.t, sim.solution.y, mesh=sim.mesh
)

plt.figure()
plt.plot(t_all, var["Measured open circuit voltage [V]"](t_all, y_all))
plt.figure()
plt.plot(t_all, var["X-averaged total heating [W.m-3]"](t_all, y_all))

# single eval fine for both
print(eval_func_a.evaluate(t, y))
print(eval_func_b.evaluate(t, y))
# Multi eval fine for a
print(eval_func_a.evaluate(t_all, y_all))
plt.figure()
plt.plot(t_all, eval_func_a.evaluate(t_all, y_all).flatten())
# Will fail here for b
print(eval_func_b.evaluate(t_all, y_all))
plt.figure()
plt.plot(t_all, eval_func_b.evaluate(t_all, y_all).flatten())
```

**Expected behaviour**
Evaluate should work for all timesteps and all variables

**Screenshots**
ValueError: all the input array dimensions for the concatenation axis must match exactly, but along dimension 1, the array at index 0 has size 1 and the array at index 2 has size 9

tinosulzer, 2020-01-17:
Yes, this is the expected behaviour at the moment. This is why you have to go via `processed_variable` to evaluate the variables one step at a time (scalar t, vector y, vector result) and then stack all the output vector into a matrix. It just makes it a lot easier to only ever think of the operations on column vectors, and not have to worry about matrices (vector t, matrix y, matrix output). There are some cases where it works by fluke because there are no difficult operations (for `a` in your example) but it isn't supported in general (any time grad, div, etc get involved)

If you really want to avoid using processed variable, you could just write your own for loop?

tinosulzer, 2020-01-17:
We should add an error message to `EvaluatorPython` if the user tries to pass in vector t / matrix y, so that at least the failure is consistent.

TomTranter, 2020-01-17:
Damn ok, was hoping i could vectorize this process as it's actually pretty slow

TomTranter, 2020-01-17:
For a simulation with 1300 spms connected in parallel using the casadi solver it takes almost 3x longer to evaluate about 10 variables individually than it does to actually solve the problem. Is there a way you could send a list of variables to evaluate to the solver and have it just spit out the current values at the end of each solve/step

tinosulzer, 2020-01-17:
Yes, that should certainly be possible. Another thing to try is converting the variables themselves to casadi before evaluating them as that might make it much faster. Can you share the script you are using (or a reduced version of it) to do this? Also, if you're not currently going via the `Solution` object to get variables, doing that might be faster as it populates a `known_evals` dictionary to avoid recalculating where possible (including when appending)

TomTranter, 2020-01-18:
Yep, thanks @tinosulzer I will update my scripts to use the new solution evals and see if we can get speed up. My thought is it will still be relatively slow as I am currently creating the python eval once and then sending multiple solutions to it in a loop and would effectively be doing the same thing with calling a method on each solution. I was hoping to concatenate all the data for the last time step and evaluate just once for each variable in each global time loop. I can set up a simplified example for us to look at at the next meeting. Also, I'd be willing to have a skype call individually soon to discuss the ecm code anyway. Just got a few things to iron out. For the variables where the eval did work "accidentally" as you put it it was a tremendous time saving so might be worth looking at in future but maybe only relevant to this use case

tinosulzer, 2020-01-19:
Ok, I see the problem. Hopefully using solution evals speeds it up but happy to schedule a call anyway. Maybe now that all of the operations are in place it would be possible to look at it again and vectorize that part.

TomTranter, 2020-01-20:
```
import pybamm
from pybamm import EvaluatorPython as ep
import numpy as np
import matplotlib.pyplot as plt
import time as record_time

pybamm.set_logging_level("INFO")

# load model
model = pybamm.lithium_ion.SPMe()

# create geometry
geometry = model.default_geometry

# load parameter values and process model and geometry
param = model.default_parameter_values
param.process_model(model)
param.process_geometry(geometry)

# set mesh
mesh = pybamm.Mesh(geometry, model.default_submesh_types, model.default_var_pts)

# discretise model
disc = pybamm.Discretisation(mesh, model.default_spatial_methods)
disc.process_model(model)

# solve model
#t_eval = np.linspace(0, 0.2, 100)

var = "Terminal voltage [V]"
# step model

time = 0

N_SPM = 20
N_steps = 10
dt = 0.1/N_steps
func = ep(model.variables[var])

solutions = [None]*N_SPM
solvers = [pybamm.CasadiSolver() for i in range(N_SPM)]

results_a = np.zeros([N_steps, N_SPM])
results_b = np.zeros([N_steps, N_SPM])
results_c = np.zeros([N_steps, N_SPM])
times = []

eval_time_a = 0.0
eval_time_b = 0.0
eval_time_c = 0.0
solve_time = 0.0

for t in range(N_steps):
    for i in range(N_SPM):
        # Solve
        st = record_time.time()
        if not solutions[i]:
            # create solution object on first step
            solutions[i] = solvers[i].step(model, dt=dt, npts=10)
        else:
            # append solution from the current step to step_solution
            solved_len = solvers[i].y0.shape[0]
            solvers[i].y0 = solutions[i].y[:solved_len, -1]
            solvers[i].t = solutions[i].t[-1]
            solutions[i].append(solvers[i].step(model, dt=dt, npts=10))
        solve_time += record_time.time() - st
        # Eval in place
        st = record_time.time()
        results_a[t, i] = func.evaluate(solutions[i].t[-1],
                                        solutions[i].y[:, -1])
        eval_time_a += record_time.time() - st
        # From solution in place
        st = record_time.time()
        results_c[t, i] = solutions[i][var](solutions[i].t[-1])
        eval_time_c += record_time.time() - st
    # Eval at end
    st = record_time.time()
    temp_t = np.asarray([solutions[i].t[-1] for i in range(N_SPM)])
    temp_y = np.asarray([solutions[i].y[:, -1] for i in range(N_SPM)]).T
    results_b[t, :] = func.evaluate(temp_t, temp_y)
    eval_time_b += record_time.time() - st
    time += dt
    times.append(time)
times = np.asarray(times)

step_voltage_a = solutions[0]["Terminal voltage [V]"]
step_voltage_b = solutions[1]["Terminal voltage [V]"]
print(np.allclose(step_voltage_a(times),
                  step_voltage_b(times)))
print(np.allclose(step_voltage_a(times),
                  results_a[:, 0]))
print(np.allclose(results_a, results_b))
print(np.allclose(results_b, results_c))
print('Solve time', solve_time)
print('Eval time individual', eval_time_a)
print('Eval time concatenate', eval_time_b)
print('Solution eval time', eval_time_c)
```

TomTranter, 2020-01-20:
Solve time 10.97115421295166
Eval time individual 0.1780099868774414
Eval time concatenate 0.01885676383972168
Solution eval time 0.8599853515625

TomTranter, 2020-01-20:
And that's one variable. The eval time grows linearly with number of variables needing to eval. Arguably his only needs to be done in the loop for variables that affect the coupled solving procedure but for me that is still quite a few as I calculate overpotentials and heat sources

TomTranter, 2020-01-20:
[image]

TomTranter, 2020-01-20:
In my ecm coupled code eval is taking longer than both the solver processes put together!

tinosulzer, 2021-01-24:
Don't think this is possible at all now we are using casadi to post-process. But is it now fast enough?

tinosulzer, 2021-01-24:
One thing we could do is post-process several variables in parallel instead of in series

tinosulzer, 2021-09-10:
I think using casadi has made post-processing fast enough, so closing

closed 2021-09-10 by tinosulzer