pybamm-team/PyBaMM #789: Events for Casadi solver when stepping

TomTranter, 2020-01-21:
**Summary**
Hi, can anyone explain why events are not implemented in the casadi solver compute_solution method and is this something that will never work or something we haven't got round to doing? There is a line saying events not implemented in this method and termination is always set to "final time". However, in the solve method when mode is safe there does seem to be some event handling. I was wondering whether this could be refactored a little so that the simulation step method could also use it as it currently just calls compute_solution.

 **Motivation**
When stepping solution it's good to know why it finished

tinosulzer, 2020-01-21:
Casadi solvers don't handle events and don't plan to (see #664). We get around this by doing a step-and-check during solve, but haven't implemented this for step. Happy to refactor `step` so that it calls step-and-check when using safe mode, as I am refactoring a lot of the solver code anyway in #784

tinosulzer, 2020-01-23:
I have this working but one problem is that it continues to step over the event after hitting it. Say on the nth iteration you hit the event, then it stops and `solver.termination = "event"`, but it can just keep going on the (n+1)th iteration because it's already gone past the event.

One option is to say "you can't step if the previous solution has hit an event". But this might cause problems with things like CCCV where you want to hit a voltage event and then keep stepping at constant voltage. Either way one can be the default and the other can be an option, the question is do we want the default to be "no stepping after events" (has to be manually disabled) or "allow stepping after events" (check for event termination has to be manually enabled)

TomTranter, 2020-01-23:
I think default should be stop when an event is met. We can string together a series of events like instructions to a potentiostat and append solutions but reset solvers

PR #800 (2020-02-04): Issue 784 simplify solver
# Description

Simplify the various solver classes by merging the `set_up` methods and getting rid of `compute_solution`. Also combined the implementations for external variables and inputs

Fixes #784 
Fixes #789 
Fixes #463 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimi [... cut]
closed 2020-02-04 by tinosulzer