pybamm-team/PyBaMM #798: User Workshop preparation email

jedgedrudd, 2020-01-24:
**Summary**
Draw up an email to explain to attendees of the User workshop what they need to do to prepare in advance.

 **Motivation**
May save installation time at the workshop.

**Additional context**
 Add any other context or screenshots about the feature request here.

tinosulzer, 2020-02-03:
### Draft installation paragraph for the email

PyBaMM is now on pip and so should be easy to install on Linux, Mac or Windows. However, to save time in the workshop, please make sure you have Python 3.6 or 3.7 (not <=3.5, not 3.8) installed. You should also know how to do the following, preferably from the command line:

- Clone or download a repository from git (to obtain the examples for the workshop)
- Install a package via pip (we strongly suggest using conda or virtual environments)

If you want to try installing PyBaMM in advance, instructions can be found in our README: https://github.com/pybamm-team/PyBaMM

### Draft timetable

Timetable seems quite un-ambitious but leaves time for people to play around with things and ask questions

10-11: Introductions, pybamm overview (show off existing capability), installation (clone or download example folder, pip install pybamm)
11-11:30: Coffee break
11:30-12:30: Running a simulation, running an experiment
12:30-13:30: Lunch
13:30-14:30: Changing settings for a simulation (from a high-level perspective of just passing options to the model or simulation)
14:30-15:00: How we plan on recognizing contributions to pybamm (papers for major releases, citations for smaller contributions)
15:00-15:30: Coffee break
15:30-16:30: Discussion and planning for v3

### Existing capability to show off

Slides to show some of the things pybamm is already being used for. Can also be used to plug upcoming papers!
- Comparing lithium-ion models, highlighting speed
- Running some of Emma Kendrick's experimental protocols
- 2+1D models @Scottmar93 / @rtimms ?
- Rate capability plots @ferranbrosa ?
- ECM coupling @TomTranter ?

### Thoughts for v3

- Coupling with MATLAB/Simulink
- Calculating sensitivities and adjoints
- Adding models for degradation and different chemistries
- ???

jedgedrudd, 2020-02-10:
Draft announcement email:
Dear Multiscale Modellers,

I am pleased to announce, on behalf of the PyBaMM team, that we will be holding a User Workshop in Oxford on Wednesday the 18th March. This is the day after the Oxford Battery Modelling Symposium (16-17th). The workshop will be held at 102-104 St Aldate's, Oxford, OX1 1BT (Map).

Sign up here.

The event is designed to train people in using PyBaMM's built-in models, with minimal coding, as well as inputting equations and parameters to suit your model and chemistry. This is primarily for MSM researchers, but please let me know if you feel that this event will be useful for a Faraday researcher not in MSM, or someone from another group entirely. We have reserved a limited number of spaces for people outside of the MSM project.

Installation Instructions:
PyBaMM is now on pip and so should be easy to install on Linux, Mac or Windows. However, to save time in the workshop, please make sure you have Python 3.6 or 3.7 (not <=3.5, not 3.8) installed. You should also know how to do the following, preferably from the command line:

•	Clone or download a repository from git (to obtain the examples for the workshop)
•	Install a package via pip (we strongly suggest using conda or virtual environments)

If you want to try to install PyBaMM in advance, instructions can be found in our README: https://github.com/pybamm-team/PyBaMM.

TomTranter, 2020-02-10:
Looks good to me - I think now we're on pip it's worth keeping master in step with this version and having a dev branch that can go ahead of it.

tinosulzer, 2020-02-10:
yes I agree this is the way to go now, after we have released v2

Scottmar93, 2020-03-10:
For the "change simulation settings" section, I am thinking of going through: 
1. how to change the c-rate (`pybamm.Simulation(model, C_rate=2)`) 
2. how to change the current to run an inbuilt drive cycle
3. how to change the parameter set to an inbuilt parameter set
4. how to change 1 parameter to a new scalar value
5. how to change 1 parameter to a new function
6. how to change grid points 
7. discuss how to input your own parameter set and drive cycles

@tinosulzer does any of this clash with your plans for "running a simulation". I was also thinking that you would cover passing options to models? 

@TomTranter I was thinking you could go over: 
8. passing in external variables
9. quick extraction of variables from pybamm
is there anything else you would like to cover? 

@tlestang with regards to point 7, is there now a function to fetch the "input" folder and put it into the current working directory when we have just pip installed pybamm? and is there any documentation on this? I found the function "pybamm_edit_parameter" within my venv/bin but struggling to figure out the workflow 

Anything else we want to cover?

rtimms, 2020-03-10:
as part of 8. it would be good to see `InputParameter`. we could loop over e.g. a diffusion coefficient then compare the results. this then hints at parameter estimation, but without the details

tinosulzer, 2020-03-10:
Sounds good, I will focus on changing the model and experiment. I might also show how to export the solution to csv/matlab
Agree that 8 should include `InputParameter`. 
Also shouldn't 2 be a special case of 5? Similarly with C-rate, maybe introducing the two different ways of doing that (changing the parameter and passing in the option) will be confusing.

Scottmar93, 2020-03-11:
Ah yeah good idea with the exporting! 

Yeah 1. and 2. are special cases of 4. and 5. I'll cover 2. at the same time as 5. so that people can see we are just modifying a parameter. 

Yeah your point about passing in a `C_rate=` argument into simulation is fair. I'll stick to just modifying the parameters dictionary.

TomTranter, 2020-03-30:
@Scottmar93 looks good to me

closed 2020-04-02 by tinosulzer