pybamm-team/PyBaMM #799: Jupyterhub launch email

jedgedrudd, 2020-01-24:
**Summary**
Draw up an email to announce the launch of our Jupyterhub and give starter information about how to use it.

 **Motivation**
Since the VM is now running, we may as well make people aware that the service is there. (addendum: please turn off the VM until the email is sent? :) 

**Additional context**
 Add any other context or screenshots about the feature request here.

martinjrobins, 2020-01-27:
VM is currently off (I turned it off on the 20th Jan)

jedgedrudd, 2020-02-10:
VM can now be turned back on - azure account is on a monthly subscription

jedgedrudd, 2020-02-10:
Can we now send out a launch email, or is it best to wait for the workshop?

tinosulzer, 2020-06-21:
I don't think we need jupyter hub now that google colab is working

tinosulzer, 2020-07-09:
Closing this issue. Have we turned off the jupyter hub?

closed 2020-07-09 by tinosulzer