pybamm-team/PyBaMM #803: After install and run-tests, I have following errors.

gyouhoc, 2020-02-05:
**Describe the bug**

I have following error after the run-tests. 

[image]

**To Reproduce**
Steps to reproduce the behaviour:

run-tests

**Expected behaviour**
no error

**Screenshots**
Provided

**Additional context**
Can I move on working with the simulator with this error?

tinosulzer, 2020-02-05:
I think this is because you're missing graphviz, try on linux
```
sudo apt-get install graphviz
```
or on mac
```
brew install graphviz
```

tinosulzer, 2020-02-05:
This test isn't that important though, and won't affect running the simulator

tinosulzer, 2020-02-06:
hi @gyouhoc , did you manage to get this working?

gyouhoc, 2020-02-06:
I haven't installed graphviz yet. I will let you know after I install it. 
However, this error doesn't influence on the PyBaMM simulation.
I am working on the tutorials now.

gyouhoc, 2020-02-07:
I installed graphviz 0.13.2 but I still have a same error.

tinosulzer, 2020-02-07:
I have opened a pull request to make graphviz an optional dependency. In the meantime, what OS are you on? And can you try with a more recent version of graphviz (e.g. 2.42.2)?

gyouhoc, 2020-02-09:
I am using Windows 10. 
I will check the most recent version of graphviz.

PR #810 (2020-02-12): #803 make visualise fail only raise logger error
# Description

Make visualise raise a logger error only if graphviz is not installed (so that graphviz is an optional dependency)
Fixes #803 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix [... cut]
closed 2020-02-12 by tinosulzer