pybamm-team/PyBaMM #816: Model Parameter data extraction

gyouhoc, 2020-02-11:
Hi, maybe you already have this feature in this program. If you have this, please direct me to the description page. 

**Summary**

Some model parameters such as electrolyte diffusivity change with time with a pre-defined function. 
Do you have a feature to have this data w.r.t. time after the simulation? So that I may complete the data table as below? 

[image]

Would you let me know any quick method for this? One thing I can do is calculating the electrolyte diffusivity with the simulation output data such Electrolyte concentration and cell temperature. 

 **Motivation**
 Why are we doing this? What use cases does it support? What is the expected outcome?

To complete the data table

**Additional context**
 Add any other context or screenshots about the feature request here.

tinosulzer, 2020-02-11:
See code to do this below (requires latest version of master). Note diffusivity varies as a function of both space and time so I'm not sure exactly what you'd be looking for in that table - but you can do min, average, max for example
```python3
import pybamm
import numpy as np

# Create model and add diffusivity to the variables
# with thermal effects to see a change in average diffusivity as temperature changes
model = pybamm.lithium_ion.DFN({"thermal": "x-full"})

# Average diffusivity (note c_e_av is constant by conservation of mass)
c_e_av = model.variables["X-averaged electrolyte concentration [mol.m-3]"]
T_av = model.variables["X-averaged cell temperature [K]"]
# all parameters can be found in pybamm.standard_parameters_lithium_ion
model.variables[
    "Average electrolyte diffusivity [m2.s-1]"
] = model.param.D_e_dimensional(c_e_av, T_av)

# Distributed diffusivity as function of x and t (note this can't be saved to csv)
c_e = model.variables["Electrolyte concentration [mol.m-3]"]
T = model.variables["Cell temperature [K]"]
model.variables["Electrolyte diffusivity [m2.s-1]"] = model.param.D_e_dimensional(
    c_e, T
)
# Min and max diffusivity
model.variables["Minimum electrolyte diffusivity [m2.s-1]"] = pybamm.min(
    model.param.D_e_dimensional(c_e, T)
)
model.variables["Maximum electrolyte diffusivity [m2.s-1]"] = pybamm.max(
    model.param.D_e_dimensional(c_e, T)
)

sim = pybamm.Simulation(model)
sim.solve(t_eval=np.linspace(0, 0.15))

# save required outputs
# you can only save outputs that only depend on time to csv
sim.solution.save_data(
    "outputs.csv",
    [
        "Time [s]",
        "Terminal voltage [V]",
        "Average electrolyte diffusivity [m2.s-1]",
        "Minimum electrolyte diffusivity [m2.s-1]",
        "Maximum electrolyte diffusivity [m2.s-1]",
    ],
    to_format="csv",
)

# spatially varying outputs can be saved to other formats, or just printed
sim.solution.save_data(
    "outputs.mat",
    ["Time [s]", "Terminal voltage [V]", "Electrolyte diffusivity [m2.s-1]"],
    to_format="matlab",
)
print(sim.solution["Electrolyte diffusivity [m2.s-1]"].entries)
```

gyouhoc, 2020-02-12:
Hi,

Thank you for your solution. This is what I need. 

However, I have following error. Does this indicate that the master is not latest one? 
If then, how do I update that? Do I pip upgrade? Reinstall? 

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-46-aec4a23f047c> in <module>
     10         "Maximum electrolyte diffusivity [m2.s-1]",
     11     ],
---> 12     to_format="csv",
     13 )

TypeError: save_data() got an unexpected keyword argument 'to_format'

tinosulzer, 2020-02-12:
Yes you need to update, how you do that depends on how you installed in the first place - if you cloned the git repository you can do `git pull`, if you downloaded the compressed file you will need to redownload a newer version. In both cases you should then run the pip installation again. Sorry it's a bit complicated - we are working on changes to make it easier!

gyouhoc, 2020-02-12:
It works very well. 

Thank you

tinosulzer, 2020-02-12:
Great! Closing

closed 2020-02-12 by tinosulzer