pybamm-team/PyBaMM #822: Cheatsheet for the workshop

ferranbrosa, 2020-02-12:
**Summary**
I think it would be very helpful for the workshop to provide a PyBaMM cheatsheet with the basic commands and syntax, so people don't have to spend too much time looking it up through the examples.

 **Motivation**
Ideas on what to include:

- How to list all the processed variables `list(model.variables.keys())`
- How to run a simulation and what are the optional parameters to pass.
- The syntax of the experiments.

**Additional context**
 Add any other context or screenshots about the feature request here.

Scottmar93, 2020-02-17:
Add to the documentation (also have a print to pdf option)

rtimms, 2020-02-29:
it might be useful to think about making some feedback forms so we get useful and constructive feedback about things to improve. if we can be specific and brief we might get better responses

rtimms, 2020-02-29:
any input for the cheat sheet welcome! then we can put something together over the next 2 weeks

ferranbrosa, 2020-03-01:
I think that, as a starting point for the upcoming workshop, we could go with how to run a simulation with everything as default, and then how to change every single aspect (in terms of what to pass and so on). Also, how to get the list of variables of the model and how to use QuickPlot.

Later on we can think of a cheatsheet with more advanced features like how to build new models.

tinosulzer, 2020-03-01:
Also how to define an experiment class, some of the parameters you might want to update more regularly (current functions, initial conditions), some of the more common variable names

tinosulzer, 2020-03-05:
Useful links for anyone wanting to contribute further (CONTRIBUTING.md, online docs)

rtimms, 2020-03-11:
I've started a CHEATSHEET.md. Please add/remove/edit/reformat as you wish

tinosulzer, 2020-03-11:
This is the collaborative markdown thing I was talking about, might make it easier to write collaboratively: https://hackmd.io/@tinosulzer/HyeLXOUH8/edit

rtimms, 2020-03-11:
ah thanks! I had forgotten you'd mentioned that

tinosulzer, 2020-05-06:
Closing this; to be edited as part of the website

closed 2020-05-06 by tinosulzer