pybamm-team/PyBaMM #850: Extra solver output

rtimms, 2020-02-21:
it would be good to optionally get some extra solver output, e.g. from casadi we can readily get something like 
```
Number of steps taken by SUNDIALS: 394
Number of calls to the user’s f function: 716
Number of calls made to the linear solver setup function: 39
Number of error test failures: 4
Method order used on the last internal step: 3
Method order to be used on the next internal step: 3
Actual value of initial step size: 2.56788e-08
Step size taken on the last internal step: 0.000225891
Step size to be attempted on the next internal step: 0.000225891
Current internal time reached: 0.000225891
Number of nonlinear iterations performed: 711
Number of nonlinear convergence failures: 3
```

tinosulzer, 2020-04-22:
Will add more general options to pass extra options to solvers, which is solver-specific

tinosulzer, 2020-04-22:
If we allow passing extra options as the dictionary, should we also make rtol and atol explicit options?
For example, in the scipy solver, should rtol and atol be set as 
```python3
ScipySolver(rtol=1e-6, atol=1e-6)
```
or 
```python3
ScipySolver(extra_options={"rtol": 1e-6, "atol": 1e-6})
```
or have the option for either?

rtimms, 2020-04-22:
I think things that are common to all solvers and are used fairly regularly (e.g. tolerances, root method) should be explicit options, and things that are specific to a particular solver or are a bit more technical should be extra options

PR #968 (2020-04-28): Issue 850 solver options
# Description

- Allow passing extra options to solvers (Fixes #850)
- Allow symbolic inputs in initial conditions (Fixes #967)
- Reuses algebraic solvers to calculate initial conditions

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up th [... cut]
closed 2020-04-28 by tinosulzer