pybamm-team/PyBaMM #857: Changes in a dependency

kinnala, 2020-02-25:
I am the author and maintainer of scikit-fem. I have noted your use of the library from which I'm grateful and wanted to give you a heads up on the upcoming 1.0.0 release. Our first stable release is getting close and there will be major internal changes in the library.

I have attempted to verify that the tests of PyBaMM run properly against the upcoming version. Some features will begin outputting a DeprecationWarning but we don't have any immediate plans to drop the support and have taken care to maintain backwards-compatibility in most of the cases.

I'm happy to help fixing things in case anything breaks or you wish to start using the new features. I expect the release to happen before summer.

tinosulzer, 2020-02-26:
Hi @kinnala , thanks for the heads up. I've just bumped up scikit-fem to 0.4.0 locally and that all looks fine. Once 1.0.0 is out we can work through getting rid of the DeprecationWarnings and making the most of the new features.

I've just realised we don't generate a citation when running the parts of the code that use scikit-fem. Is there a bibtex entry we can cite, or just the Zenodo release?

tinosulzer, 2020-02-26:
Have found the bibtex button on Zenodo 👍

kinnala, 2020-04-24:
A status update: scikit-fem 1.0.0 was released some days ago. I'm currently working on our documentation to include the changes.

I tried running pybamm test suite against 1.0.0 and, at least to me, everything seemed to work just fine (if you ignore the deprecation warnings). So no need to hurry and update anything yet. There might be some rough edges found and small fixes done in the upcoming months.

The main reason for the deprecation warnings is that the syntax for defining the forms has improved: in future more general PDE's can be defined with (hopefully) more intuitive syntax. Pybamm most likely doesn't use these new features too much but according to my preliminary tests 1.0.0 could be slightly faster due to some internal simplifications.

Currently there is a wrapper in place which fairly well reproduces the pre-1.0.0 behavior and I plan to support that for quite a some time. In future I could even provide you with a PR that fixes the warnings.

> I've just realised we don't generate a citation when running the parts of the code that use scikit-fem. Is there a bibtex entry we can cite, or just the Zenodo release?

Currently there are no other publications so Zenodo is fine.

tinosulzer, 2020-06-22:
@kinnala we are having to deal with updated dependencies now so I guess it makes sense to take care of this one too. The only warning I am seeing is for `@skfem.bilinear_form` and `@skfem.linear_form`. What should those be replaced by?

kinnala, 2020-06-22:
The new forms are defined via `skfem.BilinearForm` and `skfem.LinearForm`.

For example, this form
```python
@skfem.bilinear_form
    def mass_form(u, du, v, dv, w):
        return u * v
```
would be replaced by
```python
@skfem.BilinearForm
    def mass_form(u, v, w):
        return u * v
```

In case derivatives are needed, e.g. here
```python
@skfem.bilinear_form
    def stiffness_form(u, du, v, dv, w):
        return sum(du * dv)
```
the following should work:
```python
@skfem.BilinearForm
def stiffness_form(u, v, w):
    return sum(u.grad * v.grad)
```

Similarly for the linear forms, e.g.
```python
@skfem.linear_form
def integral_form(v, dv, w):
     return v
```
transforms to
```python
@skfem.LinearForm
def integral_form(v, w):
     return v
```

This line I find a bit confusing: https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/spatial_methods/scikit_finite_element.py#L155

I'd expect it to read
```python
@skfem.bilinear_form
def gradient_dy(u, du, v, dv, w):
    return du[0] * v
```
instead of
```python
@skfem.bilinear_form
def gradient_dy(u, du, v, dv, w):
    return du[0] * v[0]
```
I have to think what is the consequence.

kinnala, 2020-06-22:
> This line I find a bit confusing: https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/spatial_methods/scikit_finite_element.py#L155
>
> I'd expect it to read
>
> ```python
> @skfem.bilinear_form
> def [...]

Suprisingly they seem to give the same result for linear elements. I'm not exactly sure why this is the case, but my guess is that it's a lucky shot in some low-level implementational detail related to indexing.

Anyhow, I'd replace this:
```python
@skfem.bilinear_form
def gradient_dy(u, du, v, dv, w):
    return du[0] * v[0]
```
with this:
```python
@skfem.BilinearForm
def gradient_dy(u, v, w):
    return u.grad[0] * v
```

tinosulzer, 2020-06-23:
Thanks for the tips. Any ideas for this error? 
```
File "/Users/vsulzer/Documents/Energy_storage/PyBaMM/pybamm/spatial_methods/scikit_finite_element.py", line 198, in laplacian
    stiffness_matrix = self.stiffness_matrix(symbol, boundary_conditions)
  File "/Users/vsulzer/Documents/Energy_storage/PyBaMM/pybamm/spatial_methods/scikit_finite_element.py", line 276, in stiffness_matrix
    stiffness = skfem.asm(stiffness_form, mesh.basis)
  File "/Users/vsulzer/Documents/Energy_storage/PyBaMM/PyBaMM-env/lib/python3.7/site-packages/skfem/assembly/__init__.py", line 64, in asm
    return form.assemble(*args, **kwargs)
  File "/Users/vsulzer/Documents/Energy_storage/PyBaMM/PyBaMM-env/lib/python3.7/site-packages/skfem/assembly/form/bilinear_form.py", line 72, in assemble
    data[ixs] = self._kernel(u.basis[j], v.basis[i], w, dx)
  File "/Users/vsulzer/Documents/Energy_storage/PyBaMM/PyBaMM-env/lib/python3.7/site-packages/skfem/assembly/form/bilinear_form.py", line 78, in _kernel
    return np.sum(self.form(*u, *v, w) * dx, axis=1)
TypeError: stiffness_form() missing 2 required positional arguments: 'dv' and 'w'
```

kinnala, 2020-06-23:
Each bilinear form should take now 3 parameters: u, v, and w. Could it be
that you have an extra parameter? Can you push to a public branch so I can
take a look?

tinosulzer, 2020-06-23:
pushed

kinnala, 2020-06-23:
I'm on a mobile phone now but can post a PR later when I get home.

In short, you need to change the arguments to either (u, v, w) in bilinear
forms or (v, w) in linear forms.

E.g.,

def mass_form(u, du, v, dv, w):

to

def mass_form(u, v, w):

tinosulzer, 2020-06-23:
Thanks, so should we replace `du` with `u.grad` everywhere?

kinnala, 2020-06-23:
Yeah. Previously u.grad was passed to the form (function) as du and v.grad
as dv.

closed 2020-06-24 by tinosulzer

kinnala, 2020-06-24:
You might want to also update `requirements.txt` and similar lists to
```
scikit-fem >= 1.0.0
```

tinosulzer, 2020-06-24:
oops, just merged, will update in a separate PR

PR #1074 (2020-06-24): Issue 857 scikit fem
# Description

Update scikit fem method to get rid of deprecation warnings. Thanks to @kinnala for the tips

Fixes #857 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Ke [... cut]