pybamm-team/PyBaMM #867: Update summary slides

rtimms, 2020-02-29:
We link to some summary slides in the README that are a bit outdated. It would be good to update these to show off new functionality and ensure the syntax is up to date. It might also be nice to include examples/case studies of how pybamm is being used. We can probably adapt the slides we’ll be using for the MSM meeting.

tinosulzer, 2020-02-29:
Sounds good, we could just use the `pybamm_summary_slides` repo as a collection of all the presentations/papers/posters we write about pybamm? e.g. 2019-07-talk.pdf, 2020-02-paper.pdf, 2020-03-poster.pdf, etc. ?
Or just have the most recent information available?

tinosulzer, 2020-03-05:
@rtimms  can you add Jon's slides to https://github.com/pybamm-team/pybamm-supporting-material?

closed 2020-04-23 by tinosulzer