pybamm-team/PyBaMM #871: Improvements to `QuickPlot`

tinosulzer, 2020-03-04:
`QuickPlot` could do with a few updates. Feel free to add more to this list:

- More obvious error when two variables have different dimensions
- More obvious error when axlims are inf
- Allow plotting a single 3D variable (contour plot)
- Make the x-axis label "x" when the spatial scale isn't found (or raise error?)
- Should there be default output variables? Or should they always have to be passed in (which then makes it more obvious how to change them?)
- Try to clean up `set_output_variables` to make it more obvious what is being done

Scottmar93, 2020-03-04:
My thoughts are: 
1. Agree
2. Agree 
3. Yes we really need this. 
4. If spatial scale not found then raise error 
5. Yes defaults are good for pybamm.QuickPlot
6. Okay 
7. Okay 
8. Sure. Maybe just make quickplot a method of simulation that uses the default variables. 

Some additional ideas that are non essential but might be nice:
- Plot all variables in dimensional form?
- when plotting multiple models, use dashes and dots for the lines
- if passing in multiple models, allow labels for each model to also be passed in
- allow linestyles/colors to be passed for each model
- add a figsize argument 

Another thought though this really is just extra. It might be nice to have helper functions in simulation:
- slider plot (for what you describe above)
- quickplot (slider with defaults)
- Plot (just accepts 2 variables and plots them against each other)
- Plot2D (takes in 3 variables and produces the contour plot)

rtimms, 2020-03-05:
My thoughts:

- I'm also for having default variables, as it is nice just to get a quick overview without worrying about the details. 
 - I think the class should still be `QuickPlot`, as you can just call `plot` to get a static plot at a fixed time. happy for `dynamic_plot` to be renamed `slider_plot`. Maybe there could be some default collections for plots people use a lot, but this doesn't necessarily need to be done in this issue and may be best for users to implement themselves
- For the default li-ion some of the variables and dimensional and some aren't, so it would be to make this consistent 
- It looks like labels, linestyles and colors are in and can be user supplied. the default linestyles aren't looped over as it looks like there is a typo in line 339: `linestyles[j]` should be `linestyles[i]`. adding a figsize would be nice
-  maybe by default the time should be in seconds (or in seconds if the simulation is less than e.g. 2hrs). I think it is easier to read off x seconds rather than 0.x hours. maybe you could even pass how you want time to be represented e.g. seconds, hours, discharge capacity 
- it would be nice to have the option to pass an array of times to `plot` to get a single figure with profiles at the variables at different time

tinosulzer, 2020-03-05:
All sound like good points. re: linestyles, the current functionality is that you can plot several variables on the same plot and they will be differentiated by the linestyles, which is why the different solutions are only differentiated by colour. 
For example
```python3
pybamm.QuickPlot(
    solution,
    output_variables=[
        "Negative particle surface concentration",
        "Positive particle surface concentration",
        ["Electrolyte current density", "Electrode current density"],
        "Terminal voltage [V]",
    ],
)
```
gives
[image]

rtimms, 2020-03-05:
ah nice! I didn't know you could nest lists in that way to use in `QuickPlot`

Scottmar93, 2020-03-10:
When we call ipywidgets within simulation for slider plots in jupyter notebook, I think we should also pass the argument `continuous_update=False` as this seems to give more responsive behaviour :)

tinosulzer, 2020-03-10:
Are you thinking to pass this in from the notebook or detect that it is being run from a notebook?

Scottmar93, 2020-03-11:
Was thinking to just change line 536 of simulation to: 
```
widgets.interact(
                plot.plot,
                t=widgets.FloatSlider(min=0, max=plot.max_t, step=0.05, value=0),
                continuous_update=False
            )
```

Scottmar93, 2020-03-11:
Regarding linestyles, that is really neat and I didn't know you could do that either. I still think it would be nice to do models by linestyle though (if multiple variables are not plotted on the same figure) since I would use this more frequently.

rtimms, 2020-03-11:
can we just move `isnotebook` to `QuickPlot` so we can use it directly in notebooks without having to do the slider widget manually?

Scottmar93, 2020-03-11:
One additional thing that would be nice, is if QuickPlot can accept solution objects and simulation objects. If a simulation object is passed, it just does simulation.solution automatically. I am finding myself writing [sim1.solution, sim2.solution, etc] or having to do [sim.solution for sim in [sim1, sim2, sim3]] quite a bit, which is just messy

tinosulzer, 2020-03-12:
- [ ] put in the colorbar
- [ ] fix axis limits to be at 0 and x_max
- [ ] dashed lines for subdomain boundaries
- [ ] update time unit in labels
- [ ] update spatial unit in labels

PR #886 (2020-03-22): Issue 871 quickplot
# Description

Improvements to QuickPlot as discussed in #871 
Might be good to check out the branch locally and try it to assess the changes

Fixes #871 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking c [... cut]
closed 2020-03-22 by tinosulzer