pybamm-team/PyBaMM #889: ParameterValues inherit from dict

Scottmar93, 2020-03-14:
If we make ParameterValues inherit from dict, then we can tab complete to find parameter values within ipython (and so Jupyter notebooks). N.B tab complete already works with model.variables in ipython and jupyter notebooks

tinosulzer, 2020-03-14:
I removed the inheritance from dict because it was causing problems with pickle and I read somewhere that it was better to define your own `__getitem__`, etc anyway. Happy to go back to dict if you can get it passing all the tests. Otherwise, it might be easier to just add a `__dir__` or `_ipython_key_completions_()` method as suggested [here](https://ipython.readthedocs.io/en/stable/config/integrating.html)

Scottmar93, 2020-03-14:
Ah cheers, will look into the key completions :) thought this was just a simple thing we missed.

PR #893 (2020-03-18): Issue 889 tab complete parameter values
# Description

Added `_ipython_key_completions_()` function to parameter values.

Fixes #889 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [X] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checkl [... cut]
closed 2020-03-18 by Scottmar93