pybamm-team/PyBaMM #892: Update Windows installation instructions

ferranbrosa, 2020-03-17:
The Windows installation instructions still suggest using WSL. We should add the standard Windows installation instructions (I would still keep those, but aimed maybe to more advanced users). Also, @rtimms suggested double-checking the other instructions.

tlestang, 2020-03-19:
The instructions will change rather considerably following #897 .
For Windows, the WSL isn't needed anymore, and I can update the instructions as part of #897 
There's also a conda package coming

ferranbrosa, 2020-03-21:
Great, so we can address all of them together. I helped a colleague at Birmingham installing it so I had to write the steps anyway. I will put them below in case they help. I think we should keep in mind that the instructions might be run by users with very little knowledge of the use of the terminal, that is why I kept the instructions at a high level of detail.

You first need to install Python from their website (https://www.python.org/downloads/). Make sure to install a 3.6 or 3.7 version, as PyBaMM does not work in 3.8 versions. You might need admin rights for this. Make sure to tick the box "Add Python 3.X to PATH". 

Now you will need to use the terminal. If you type `cmd` in the Windows search bar that will lead you to the Command Prompt. The basic commands there are (here there is a cheatsheet with examples http://www.cs.columbia.edu/~sedwards/classes/2015/1102-fall/Command%20Prompt%20Cheatsheet.pdf):
	• `cd` followed by the name of a directory to move to that directory.
	• `dir` to show the contents of that directory.
	• `mkdir` followed by the name of a directory to create that directory.

When you launch the terminal you will be in your home directory, so you can just create a directory there called Pybamm doing:
```
mkdir PyBaMM
```
And go into the new directory
```
cd PyBaMM
```

Now we create a virtual environment in which we will install PyBaMM. You have to type
```
python -m venv env
```

To activate this environment you should type
```
env\Scripts\activate.bat
```
You will know the environment is active as (env) appears at the beginning of the command line.

You will have to activate the environment every single time you start the command line and want to use PyBaMM. If you want to deactivate it you just type
```
deactivate
```

Now, having the virtual environment activated you can just type
```
pip install pybamm
```
to install PyBaMM. This will take a few minutes.

PR #926 (2020-03-30): Issue 892 install windows
# Description

Added basic Windows installation instructions (very basic, just for the workshop). I kept the WSL instruction for more advanced users (maybe they can be removed, but I find WSL quite useful).

Fixes #892

## Type of change

Minor change in the documentation, so none of the points below applies.

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New fe [... cut]
closed 2020-03-30 by tinosulzer