pybamm-team/PyBaMM #894: Model options class

Scottmar93, 2020-03-18:
**Summary**
Create a class to set model options with helper methods such as `info` to view the current set of model options. 

 **Motivation**
Makes it easier to set model options without having to look through the documentation. Also creates a natural place in the documentation for putting model options (instead of base_battery_model).

tinosulzer, 2020-06-15:
What functionality does this add compared to the dictionary printed by `model.options`?

tinosulzer, 2021-02-03:
@Scottmar93 are you still wanting to be assigned to this?

Scottmar93, 2021-02-06:
Have unassigned myself. I imagine the code might have changed quite a bit since I created this issue so might no longer be relevant.

tinosulzer, 2021-02-06:
I think it would still be useful to have a nice way of printing the possible options

priyanshuone6, 2021-02-25:
Hey, could you please elaborate more on this issue?

tinosulzer, 2021-02-26:
Currently model.options is a dictionary and the `BaseBatteryModel` class takes care of:
- providing docstrings for the available options
- checking the submitted options are possible (both alone and in combination with others)

Instead, `model.options` could overload the dict class (`Options` class) to do these things itself:
- have `Options.print_options()` print the possible options with the ones currently selected emphasized
- have `Options.print_detailed_options()` print what is currently the docstring for `model.options`
- have an underlying dict of available options e.g.
```python3
{"surface form": ["false", "differential", "algebraic"], 
"thermal": ["isothermal", "lumped", "x-lumped", "x-full"],
}
```
and then in `Options.update` check that the requested update for an option is part of the available options (basically what `@options.setter` in `BaseBatteryModel` does already but without needing to do each case manually)

priyanshuone6, 2021-03-04:
@tinosulzer Sorry for the delay, I was brushing up my basics on classes and objects.
I am trying to modularize options like
```
class Options(pybamm.FuzzyDict):
    """docstring"""
    def __init__(self, extra_options):
        default_options = {
            "operating mode": "current",
            "dimensionality": 0,
            "surface form": "false",
            "convection": "none",
                  .
                  #code
                  #code
                  .

        super().__init__(extra_options)
```
and in `BaseBatteryModel` class
```
    @options.setter
    def options(self, extra_options):
        self._options = Options(extra_options)
```
but I observed that the options code uses `self` which is linked to the class `BaseBatteryModel` in various places like
https://github.com/pybamm-team/PyBaMM/blob/900e52ea1e5c391a977dc22a69f6df4d6b56144c/pybamm/models/full_battery_models/base_battery_model.py#L294-L296
https://github.com/pybamm-team/PyBaMM/blob/900e52ea1e5c391a977dc22a69f6df4d6b56144c/pybamm/models/full_battery_models/base_battery_model.py#L307-L308
https://github.com/pybamm-team/PyBaMM/blob/900e52ea1e5c391a977dc22a69f6df4d6b56144c/pybamm/models/full_battery_models/base_battery_model.py#L326-L329
If I move this code in the `Options` class, the code won't work. We should think about redesigning it in a way that `BaseBatteryModel` is not writing code for its subclasses.

tinosulzer, 2021-03-04:
Probably the easiest solution is to leave the lines you linked in the `options.setter` function in BaseBatteryModel, i.e. outside the Options class

PR #1418 (2021-03-10): Options class in base_battery_model
# Description
Create class Options to modularize model options

Fixes #894 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/master/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style i [... cut]