pybamm-team/PyBaMM #904: Create PyBaMM conda package

tlestang, 2020-03-20:
**Summary**
Make PyBaMM available as a conda package, for instance through [conda-forge](https://conda-forge.org/#about).
```
$ conda install -c conda-forge pybamm
```
I opened a PR to add  PyBaMM to conda-forge, see [conda-forge/staged-recipes#11046](]https://github.com/conda-forge/staged-recipes/pull/11046). It's been reviewed quickly but now I've been waiting for the merge for a week. This should happen soon I hope :)

 **Motivation**
The Anaconda and miniconda python distributions are widely used. Although the `conda` package
manager now integrates fairly well with `pip`, installing PyBaMM through `pip` within an Anaconda environment may lead to dependencies issues and [is discouraged](https://www.anaconda.com/using-pip-in-a-conda-environment/).
Furthermore, providing a conda package for PyBaMM would give Anaconda (or miniconda) users the possibility to install PyBaMM as they install any other conda package.

**Additional context**
conda packages are built using [conda-build](https://docs.conda.io/projects/conda-build/en/latest/index.html), from a `meta.yaml` (also known as a _recipe_) file that describes the package: its source code, how to install and test it, and some metadata. 
These packages can technically be hosted and distributed from anywhere, but [conda-forge](https://conda-forge.org/#about) serves as a central, community-driven repository for conda packages (a bit like PyPI for regular Python packages). 
To contribute a package to conda-forge, one must fork the [conda-forge/staged-recipes](https://github.com/conda-forge/staged-recipes) repo, add the new recipe(s), and open a PR for the original repo. Upon the merge, the package is built and uploaded to [anaconda.org/conda-forge](https://anaconda.org/conda-forge).

**Dependencies**
When building a conda package, all of *its dependencies must also be available as conda packages*.
Between the anaconda and conda-forge channel, most of pybamm's dependencies are statisfied, except **anytree**, **scikit-fem** and **casadi**.
I built conda packages for anytree and scikit-fem, so these should also be available on conda-forge and therefore as PyBaMM dependencies once my [PR](]https://github.com/conda-forge/staged-recipes/pull/11046) is merged.
The case of **casadi** remains only problematic for windows support, as casadi is already available on conda-forge but only for linux and macos. I considered building a conda package for casadi for Windows but quickly realised that it is not at all straightforward (this would require compiling a lot of C and the dependency chain looks fairly long!). Actually compiling casadi on windows [is strongly discouraged](https://github.com/casadi/casadi/wiki/InstallationWindows).
It's a shame because Windows support was actually one of my main motivation for providing pybamm as a conda package.

schatt89, 2020-06-22:
Hello! I tried to install pybamm in a conda environment, the installation went fine but I couldn't load the library with the following error:

`cannot import name 'UndefinedOperationError' from 'pybamm.expression_tree.exceptions' (/Users/anna/miniconda3/envs/main/lib/python3.7/site-packages/pybamm/expression_tree/exceptions.py)`

I'm not sure what that means, can you please help me to fix it?

tlestang, 2020-06-24:
Hello @schatt89 ! Thanks for reporting that issue. Could you please describe how you installed PyBaMM? Particularly, did you clone the repository from GitHub? Did you get the conda package from conda-forge?

I haven't had a look at the conda package for PyBaMM package in a while, and it's not up to date. Until the conda package is fixed, I suggest you use `pip` inside a specific conda environment, _i.e_
```
$ conda deactivate # make sure no conda env is currently activated
$ conda create --name pybamm python=3.6
$ conda activate pybamm
$ (pybamm) pip install pybamm
```

tlestang, 2020-06-25:
Could you paste the full error message ? It looks like something I thought had been fixed a few weeks ago.

Assuming that the error message states that some component of the suitesparse library can't be found, make sure that the sundials library is installed:
```
brew install sundials
```
This step is required on macOS (see [install notes](https://pybamm.readthedocs.io/en/latest/#using-pip))

tlestang, 2020-06-26:
For clarity, let me just amend the installation steps above for specifically for macOS (and conda) users:
```
$ brew install sundials
$ conda deactivate # make sure no conda env is currently activated
$ conda create --name pybamm python=3.7 # (or 3.6 if preferred. Note that 3.8 isn't currently supported)
$ conda activate pybamm
$ (pybamm) pip install pybamm
```

The only difference with the above is the addition of the `brew install sundials` command.

schatt89, 2020-06-29:
I installed `sundails` and now everything works fine, thank you. Sorry for the slow response!

tinosulzer, 2021-02-04:
Could [poetry](https://python-poetry.org/) help with this somehow?

tlestang, 2021-02-05:
I don't think poetry can build conda packages, is that what you mean?
PyBaMM is available with conda now, just without the idaklu solver. 
This probably should be a separate issue, and this one can be closed.

tinosulzer, 2021-02-05:
Oh I thought I read it could, never mind. Not a big deal about the idaklu solver, will close this one.
Also not available through conda on windows, right?

closed 2021-02-05 by tinosulzer